greatscottgadgets/hackrf #579: firmware:git-90d4014  and only one byte per sample

lunakk-PL, 2019-01-19:
My HackRF One was running soft ver 2018.xx.xx
Every thing was perfect until today.
I decide to update firmware from gitgub.
Update went smooth
my hackfr is alive and running

hackrf_info version: git-90d4014
libhackrf version: git-90d4014 (0.5)
Firmware Version: git-90d4014 (API:1.03)

problem is i get only one byte per sample:

$ hackrf_transfer -r /dev/null
call hackrf_set_sample_rate(10000000 Hz/10.000 MHz)
call hackrf_set_hw_sync_mode(0)
call hackrf_set_freq(900000000 Hz/900.000 MHz)
Stop with Ctrl-C
10.0 MiB / 1.000 sec = 10.0 MiB/second
10.0 MiB / 1.000 sec = 10.0 MiB/second
10.0 MiB / 1.000 sec = 10.0 MiB/second
^CCaught signal 2
 5.5 MiB / 0.558 sec =  9.9 MiB/second

Exiting...
Total time: 3.55832 s
hackrf_stop_rx() done
hackrf_close() done
hackrf_exit() done
fclose(fd) done
exit

call hackrf_set_sample_rate(20000000 Hz/20.000 MHz)
call hackrf_set_hw_sync_mode(0)
call hackrf_set_freq(900000000 Hz/900.000 MHz)
Stop with Ctrl-C
19.9 MiB / 1.000 sec = 19.9 MiB/second
19.9 MiB / 1.000 sec = 19.9 MiB/second
19.9 MiB / 1.000 sec = 19.9 MiB/second
^CCaught signal 2
 3.1 MiB / 0.155 sec = 20.2 MiB/second

Exiting...
Total time: 3.15584 s
hackrf_stop_rx() done
hackrf_close() done
hackrf_exit() done
fclose(fd) done
exit

### Version information
Ubuntu 18.04.1 LTS

lunakk-PL, 2019-01-19:
I got it: CPLD update requierd also :)

hackrf_cpldjtag -x hackrf/firmware/cpld/sgpio_if/default.xsvf

dominicgs, 2019-01-20:
This ties into CPLD versioning, it would be really great to notify users of a firmware-CPLD mismatch (see #565 )

closed 2019-01-20 by dominicgs