greatscottgadgets/hackrf #587: HackRF One very poor reception

EwartM, 2019-02-03:
My HackRF One suddenly has very poor reception. Local FM stations are showing up extremely weak and there are strong constant signals that don't belong in the FM band. Please see [this video](https://youtu.be/DsDpAiCQE6I)

hackrf_info version: unknown
libhackrf version: unknown (0.5)
Found HackRF
Index: 0
Serial number: 000000000000000014d463dc2f61b8e1
Board ID Number: 2 (HackRF One)
Firmware Version: 2018.01.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x00574752

Things I have tried:
1. changed Antennas
2. changed gain settings (dropping RF gain does drop signal strength by around 14db)
3. checked for dead short on board between antenna and ground
4. changed computers and reset gqrx to defaults
5. changed firmware (DFU, SPI, CPLD) to 2017 and back to 2018
6. run hackrf_transfer with various gain settings and amp on / off
7. upgraded software through apt-get

Notes:
1. Transmission of a 433MHz fob held close to the HackRF is received as expected with a very strong signal
2. Remote stations are barely received
3. There are weird, strong constant signals all over the FM band that do not belong there
4. Nothing I know of changed from the last time the device was powered except that it went through a carry on luggage x-ray machine at an airport 
5. I followed some [instructions](http://www.t4f.org/articles/repairing-the-hackrf/ )  to test the RX LNA and it looks like it passed the test

DavidVanHorn, 2019-02-04:
Internal noise and "Birdies" are a problem with all wideband receivers.  I'm about to write up something I did this weekend that might interest you.

EwartM, 2019-02-04:
Cool. Let me know when you have the link. My problem is that internal noise is all I'm hearing - outside signals are barely there.

DavidVanHorn, 2019-02-04:
https://github.com/mossmann/hackrf/issues/588

I do see a lot of notes about the front end amp being "delicate", and I'm hoping yours isn't blown.
I'm going to be leaving a 3dB pad on mine unless I have some reason to remove it.

mossmann, 2019-02-06:
In the video it looks like you are tuning the center frequency of the HackRF One to 98 MHz which is exactly the frequency of the signal you are trying to receive.  However, you also have DC blocking enabled in your software which filters out your center frequency.  Can you please try using offset tuning?
https://github.com/mossmann/hackrf/wiki/FAQ#how-do-i-deal-with-the-dc-offset

EwartM, 2019-02-08:
Thanks for the replies. Here it is with the DC offset away from the signal I'm trying to receive at 98.7MHz
[image]
Unfortunately there is no signal from the HackRF.
When I tune the a very strong FM station I might get some reception but it's barely there.
I see lots of what I assume is internal noise.
[image]
On a side note, do you know off hand of a drop in replacement for the Avago MGA-81563 amps? It would save me some time looking
Thanks

EwartM, 2019-02-10:
I'm back in an area where there are strong signals and it looks like everything is working, so I guess I was just in a very weak signal area. Thanks for the help and sorry to cost you time.

closed 2019-02-10 by EwartM