greatscottgadgets/hackrf #590: my HackRF lost power

Valent1234, 2019-02-05:
### Steps to reproduce
1.my HackRF + portapack lost power
2.
3.

### Expected behaviour
Tell us what you expect should happen

### Actual behaviour
After updating the firmware portapack-havoc-bbb5d, my hackrf does not turn on. U-23 chip is powered but nothing happens ...

### Version information 
**Operating system**:

**hackrf_info output:**

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
```
Insert any commandline or build output here
```

Valent1234, 2019-02-05:
перед тем как hackrf+portapack пропало питание, было замечено застывание дисплея после включения неоднократно.

Valent1234, 2019-02-05:
before hackrf + portapack power was lost, the display was frozen after switching on repeatedly.

Valent1234, 2019-02-05:
sorry for my English

GravesJake, 2019-02-05:
Are you able to load the latest hackrf firmware from [git](https://github.com/mossmann/hackrf)? (not havoc)

Valent1234, 2019-02-06:
Unfortunately, hackrf does not turn on at all, the computer does not detect it ...

Valent1234, 2019-02-06:
I have a suspicion that the lpc4320fbd144 chip is broken. but I don’t know how to test it ...

GravesJake, 2019-02-06:
What operating system are you using? Have you tried using [DFU mode](https://github.com/mossmann/hackrf/wiki/Updating-Firmware#only-if-necessary-dfu-boot)? Can you run the command `hackrf_info` and post the output?

mossmann, 2019-02-06:
Please try DFU mode with the Portapack removed.

(Have you seen something like this triggered by a firmware update, @sharebrained?)

Valent1234, 2019-02-06:
I turned off the portapack and the result is the same. there is no power hackrf (the lights are off), the computer does not detect the device.

I'm not claiming that a firmware upgrade led to this result ...

Valent1234, 2019-02-06:
since the computer does not connect to hackrf, I can’t use DFU mode and the hackrf_info command...

Valent1234, 2019-02-06:
Please tell me at what points on the hackrf (P ..) circuit board can you determine the health of the U23 chip?

jboone, 2019-02-06:
> (Have you seen something like this triggered by a firmware update, @sharebrained?)

I have not seen anything that made DFU mode not work.

I have seen a few instances of soldering defects on HackRF header solder joints, leading to malfunctions only when using PortaPack firmware. This would not impact the DFU function.

I have seen the battery-backed RAM (powered by the PortaPack coin battery) get into a state that crashes the PortaPack or HAVOC firmware. This can be addressed by detaching the PortaPack from the HackRF for a few seconds, long enough for the RAM in the LPC4320 to bleed down. However, this issue would not impact the DFU function.

It is possible there were code changes in the HackRF CPLD code that could trigger a soldering/hardware defect in a HackRF. I don't know of any changes that would cause this, but can't rule it out. This would only be a problem if the HackRF CPLD was not tested by boundary scan for shorts and opens, or there was a soldering issue that was made worse by mechanical or thermal stress after testing.

It sounds like the hardware did work initially, and the problem arose at the time a firmware update was performed?

@Valent1234 who did you purchase your HackRF and PortaPack from?

jboone, 2019-02-06:
> I turned off the portapack and the result is the same. there is no power hackrf (the lights are off), the computer does not detect the device.

@Valent1234 If you see *no* power lights when the HackRF is connected to USB, then there must be a short that has developed. I would re-test with different USB ports, cables, and computers, to rule those out as issues. If those changes do not fix the issue, then the HackRF power supply may have failed, or a dead short somehow now exists in the HackRF that is causing the power supply to go into a current limit mode and shut off voltage to the HackRF. Please check the voltages at the power supply, both the 1.8V and 3.3V supplies coming out of U21.

Valent1234, 2019-02-06:
voltage 1.8v and 3.3v at the output of U21 is absent.
portapack bought here:
https://ru.aliexpress.com/item/2017-PORTAPACK-HACKRF-ONE-SDR/32890461672.html?spm=a2g0s.9042311.0.0.61e433edpgfv68
Everything worked fine without comment for about a year.

Valent1234, 2019-02-06:
Thank you for your attention to my problem. Upon closer examination, the DC-DC converter (U21) is likely to burn. I will change. what could cause this is difficult to say.

mossmann, 2019-02-12:
It may have been a temporary short that caused the failure of U21.  Replacing it is likely to fix the problem, but you should subsequently test your supplies.

closed 2019-02-12 by mossmann