greatscottgadgets/hackrf #592: [NOT AN ISSUE] Need help to access an hack rf remotly

TurtleForGaming, 2019-02-10:
As said by the question is there a way for me to connect the hack rf to a raspberry like a rtl-sdr and launch a similar program like rtl_tcp to be able to use sdr# from an other computer? I've seen that the was a quick port to the hackrf but it doesn't complie now due to the `rtlsdr_set_tuner_if_gain` function being remove or renamed idk. SO if anyone can give me any clue on how I might do this I'm open

unixpunk, 2019-02-11:
Have you looked into SoapySDR+SoapyHackRF+SoapyRemote?  It currently doesn't support transport over the internet, but if you have a wifi/lan connection then it will work.  HTH

TurtleForGaming, 2019-02-11:
So after trying it out the only problem is that I only found cubicsdr and gqrx that can connect to a soapyremote server but I wanted something a bit more like sdr# / sdr-console. Neverthe less thank you it works great!

unixpunk, 2019-02-11:
I haven't actually used it, so only making guesses here, but I think SpyServer supports HackRF and can run on an Pi.  Could investigate that route as well.  Hope that helps!  Also, to note, it would be any sdr console that supports SoapySDR should also support SoapyRemote, so sdrangel should also work, in case you were interested in that one.

TurtleForGaming, 2019-02-12:
Nop spyserver support only airspys and rtl-sdr. But thanks a lot for the help

closed 2019-02-12 by TurtleForGaming