greatscottgadgets/hackrf #596: To Tranfer wrong co-ordinates from HackRFOne

devanand84, 2019-02-28:
### Steps to reproduce
1.:~/Downloads/gps-sdr-sim-master$ ./gps-sdr-sim -b 8 -e brdc3540.14n -l 46.257664,11.956105,100

2Downloads/gps-sdr-sim-master$ hackrf_transfer -t gpssim.bin -f 1575420000 -s 2600000 -a 1 -x 0

3.

### Expected behaviour
Then my gps device must show this latitude & longitude. But it is not working as expected.

### Actual behaviour
Transmit LED is ON, but no spoofing happens on GPS device

### Version information
**Operating system**:
Ubuntu 14.06

**hackrf_info output:**
Found HackRF board 0:
USB descriptor string: 0000000000000000325866e621601823
Board ID Number: 2 (HackRF One)
Firmware Version: 2018.01.1
Part ID Number: 0xa000cb3c 0x005e4768
Serial Number: 0x00000000 0x00000000 0x325866e6 0x21601823

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
```
Insert any commandline or build output here
```

GravesJake, 2019-03-08:
Are you able to do other things with your HackRF? Such as receive FM radio signals or view WiFi packets. Also be sure to do any spoofing in a shielded enclosure to comply with your local laws.

devanand84, 2019-05-01:
YES !! I am able to receive the FM radio signals. I am following the local laws & this will be just for TESTING purpose only.

GravesJake, 2019-05-21:
Sorry for the late response, I'd need more details to try and help you further. What specifically is happening?

closed 2019-11-26 by miek

febs, 2019-11-26:
It could have been be the wrong antenna length btw.