greatscottgadgets/hackrf #599: No data?

Polishfuze, 2019-03-02:
So I bought my first HackRF after having an RTL SDR dongle for a while. It arrived yesterday and I plugged in the antenna and then the HackRF into the PC. I updated the drivers with Zadig (Im on windows) and launched SDR Console V3 (3.0.5 to be exact)... There is nothing. It auto detects the HackRF One but there is no data coming through, no static in the FFT window, no noise, no flat line, NOTHING. What am I supposed to do now? I've tried to update using the updater from the wiki but it doesn't work, dump below. I also tried the receive command in GNU Radio, doesn't work, dump below. I tried using DFU mode, didn't work, dump below. Any suggestions?

Info: hackrf_info
`
hackrf_info version: git-370ee2d
libhackrf version: git-370ee2d (0.5)
Found HackRF
Index: 0
Serial number: 0000000000000000654067dc22569967
Board ID Number: 2 (HackRF One)
Firmware Version: 2018.01.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x00614362
`

Receive: hackrf_transfer -d 0000000000000000654067dc22569967 -r D:\SDR\recordings\tst
`
call hackrf_set_sample_rate(10000000 Hz/10.000 MHz)
call hackrf_set_hw_sync_mode(0)
call hackrf_set_freq(900000000 Hz/900.000 MHz)
Stop with Ctrl-C
 0.0 MiB / 1.002 sec =  0.0 MiB/second

Couldn't transfer any bytes for one second.

Exiting... hackrf_is_streaming() result: HACKRF_TRUE (1)
Total time: 1.00523 s
hackrf_stop_rx() done
hackrf_close() done
hackrf_exit() done
fclose(fd) done
exit
`

Updater:
`*** Re-flash the HackRF with HackRF firmware ***

Connect your HackRF One to a USB port on your computer.

If using a PortaPack, put the PortaPack in HackRF mode by selecting
the "HackRF" option from the main menu.

Press any key to continue . . .

Devices detected: 0 DFU, 0 HackRF
ERROR: HackRF not found.
Please connect a HackRF One to your computer's USB port.
If using a PortaPack, please put it in HackRF mode after connecting it.

Press any key to continue . . .`

DFU:
`*** Run HackRF firmware in RAM via LPC DFU ***

This is used to "unbrick" your HackRF, if you are no longer able to use
HackRF tools to flash or operate your HackRF.

Connect your HackRF One to a USB port on your computer.

Hold down both the DFU and RESET buttons on the HackRF.
Then release the RESET button (closest to the edge).
Then release the DFU button.

Press any key to continue . . .

dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

DFU suffix CRC does not match
A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 1fc9:000c
Run-time device DFU version 0100
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0100
Device returned transfer size 2048
Copying data from PC to DFU device
Download        [=========================] 100%        21992 bytes
Download done.
unable to read DFU status after completion
can't detach
Resetting USB to switch back to runtime mode

Press any key to continue . . .`

James-Brown, 2019-03-02:
Does SDR Console see the HackRF in ‘Select Radio’? 
I use Hack and SDR Console and it works fine for me.

From: Polish Fuze 
Sent: Saturday, March 2, 2019 09:25
To: mossmann/hackrf 
Cc: Subscribed 
Subject: [mossmann/hackrf] No data? (#599)

So I bought my first HackRF after having an RTL SDR dongle for a while. It arrived yesterday and I plugged in the antenna and then the HackRF into the PC. I updated the drivers with Zadig (Im on windows) and launched SDR Console V3 (3.0.5 to be exact)... There is nothing. It auto detects the HackRF One but there is no data coming through, no static in the FFT window, no noise, no flat line, NOTHING. What am I supposed to do now? I've tried to update using the updater from the wiki but it doesn't work, dump below. I also tried the receive command in GNU Radio, doesn't work, dump below. I tried using DFU mode, didn't work, dump below. Any suggestions?

Info: hackrf_info
hackrf_info version: git-370ee2d libhackrf version: git-370ee2d (0.5) Found HackRF Index: 0 Serial number: 0000000000000000654067dc22569967 Board ID Number: 2 (HackRF One) Firmware Version: 2018.01.1 (API:1.02) Part ID Number: 0xa000cb3c 0x00614362

Receive: hackrf_transfer -d 0000000000000000654067dc22569967 -r D:\SDR\recordings\tst
`
call hackrf_set_sample_rate(10000000 Hz/10.000 MHz)
call hackrf_set_hw_sync_mode(0)
call hackrf_set_freq(900000000 Hz/900.000 MHz)
Stop with Ctrl-C
0.0 MiB / 1.002 sec = 0.0 MiB/second

Couldn't transfer any bytes for one second.

Exiting... hackrf_is_streaming() result: HACKRF_TRUE (1)
Total time: 1.00523 s
hackrf_stop_rx() done
hackrf_close() done
hackrf_exit() done
fclose(fd) done
exit
`

Updater:
`*** Re-flash the HackRF with HackRF firmware ***

Connect your HackRF One to a USB port on your computer.

If using a PortaPack, put the PortaPack in HackRF mode by selecting
the "HackRF" option from the main menu.

Press any key to continue . . .

Devices detected: 0 DFU, 0 HackRF
ERROR: HackRF not found.
Please connect a HackRF One to your computer's USB port.
If using a PortaPack, please put it in HackRF mode after connecting it.

Press any key to continue . . .`

DFU:
`*** Run HackRF firmware in RAM via LPC DFU ***

This is used to "unbrick" your HackRF, if you are no longer able to use
HackRF tools to flash or operate your HackRF.

Connect your HackRF One to a USB port on your computer.

Hold down both the DFU and RESET buttons on the HackRF.
Then release the RESET button (closest to the edge).
Then release the DFU button.

Press any key to continue . . .

dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

DFU suffix CRC does not match
A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 1fc9:000c
Run-time device DFU version 0100
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0100
Device returned transfer size 2048
Copying data from PC to DFU device
Download [=========================] 100% 21992 bytes
Download done.
unable to read DFU status after completion
can't detach
Resetting USB to switch back to runtime mode

Press any key to continue . . .`

Polishfuze, 2019-03-02:
Yes, it shows up but doesn't get or display any data

James-Brown, 2019-03-02:
Have you used a simple RTL Dongle – R820T on SDR Console with success?

From: Polish Fuze 
Sent: Saturday, March 2, 2019 12:49
To: mossmann/hackrf 
Cc: James Brown ; Comment 
Subject: Re: [mossmann/hackrf] No data? (#599)

Yes, it shows up but doesn't get or display any data

Polishfuze, 2019-03-02:
No, with sdr console no, but I've used it with SDR# and the same symptoms are in SDR# when using the sdr#

James-Brown, 2019-03-03:
Did you check the ‘second’ audio output slider on SDR Console?
Its under the AGC Off portion of the DSP panel.

From: Polish Fuze 
Sent: Saturday, March 2, 2019 13:38
To: mossmann/hackrf 
Cc: James Brown ; Comment 
Subject: Re: [mossmann/hackrf] No data? (#599)

No, with sdr console no, but I've used it with SDR# and the same symptoms are in SDR# when using the sdr#

  Have you used a simple RTL Dongle – R820T on SDR Console with success? From: Polish Fuze Sent: Saturday, March 2, 2019 12:49 To: mossmann/hackrf Cc: James Brown ; Comment Subject: Re: [mossmann/hackrf] No data? (#599) Yes, it shows up but doesn't get or display any data

Polishfuze, 2019-03-03:
This is the problem I'm having, pics with all the pages so you can see them in this [Imgur Album](https://imgur.com/gallery/UsVIV3Y)

Polishfuze, 2019-03-05:
I might have figured out what has happened, I think my oscillator is dead, because I probed the CLKOUT SMA connector's center pin and it shows a steady 3.11V with a bit of noise (probably coming from my test probe) is there a better way of finding out whether the oscilator is the problem?

James-Brown, 2019-03-06:
You could install one of those external TCXO clocks.
https://www.ebay.com/itm/HackRF-One-External-Clock-GPS-Experiment-high-precision-TCXO-clock-PPM0-1/172291183638?hash=item281d5ad416:g:5ekAAOSw9NdXuwap

From: Polish Fuze 
Sent: Tuesday, March 5, 2019 13:38
To: mossmann/hackrf 
Cc: James Brown ; Comment 
Subject: Re: [mossmann/hackrf] No data? (#599)

I might have figured out what has happened, I think my oscillator is dead, because I probed the CLKOUT SMA connector's center pin and it shows a steady 3.11V with a bit of noise (probably coming from my test probe) is there a better way of finding out whether the oscilator is the problem?

miek, 2019-11-27:
Sorry for the long delay for a response. Are you still having this issue? Did you get any further with debugging the missing CLKOUT?

closed 2019-12-16 by miek