greatscottgadgets/hackrf #603: Baseband lowpass filter in osmocom source seems to have no effect

johschmitz, 2019-03-03:
### Steps to reproduce
1. In Osmocom Source I am setting the Ch0: Bandwidth(Hz) to 2e6 (sampling rate 8e6).
2. When observing the spectrum with QT GUI Frequency sink nothing seems to have changed.

### Expected behaviour
I am expecting that the "Bandwidth" option sets the baseband filter in the hardware.
Frequencies above/below +/-1MHz should be attenuated.

### Actual behaviour
Filter seems to have no effect.

### Version information
**Operating system**: Linux Mint 19

**hackrf_info output:**
hackrf_info version: unknown
libhackrf version: unknown (0.5)
Found HackRF
Index: 0
Serial number: 0000000000000000a74461c838856a9f
Board ID Number: 2 (HackRF One)
Firmware Version: git-9bbbbbf (API:1.02)
Part ID Number: 0xa000cb3c 0x005d476e

**third party software**

GNU Radio Companion 3.7.11

GravesJake, 2019-03-11:
Would you mind uploading screenshots of your frequency display with the filter on/off?

miek, 2019-11-26:
The difference won't be noticeable unless you increase the IF/BB gains, here's an example from `osmocom_fft`:

[image]
[image]

closed 2019-11-26 by miek