greatscottgadgets/hackrf #605: Hackrf not running under Ubuntu 18.04

pelixm, 2019-03-06:
### Steps to reproduce
1.Connect Hackrf
2 .Start GQRX
3.Select Hackrf device
4. Start processing

### Expected behaviour
Dongle should be activated . RX led should light . Data should arrive from the dongle. Waterfall should show signals

### Actual behaviour
Nothing of above happens.
Note 1: Under same system everything works fine if I connect to GQRX a RTL-SDR dongle.
Note 2: The HACKRF dongle looks ok because I can succesfully run it (on same computer) using Windows and Sdrsharp.

### Version information
Ubuntu 18.04
Gqrx-sdr 2.9-2
gr-osmosdr 0.1.4-14
Hackrf 2018.01.1-2

**hackrf_info output:**

mauro@mauro-Latitude-D630:~$ hackrf_info
hackrf_info version: unknown
libhackrf version: unknown (0.5)
Found HackRF
Index: 0
Serial number: 0000000000000000644064dc214715cd
Board ID Number: 2 (HackRF One)
Firmware Version: 2018.01.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x00604355

### Output
```
mauro@mauro-Latitude-D630:~$ gqrx
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

Controlport disabled
No user supplied config file. Using "default.conf"
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.11
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
FM demod gain: 3.05577
Resampling audio 96000 -> 48000
IQ DCR alpha: 1.04166e-05
Using audio backend: auto
BookmarksFile is /home/mauro/.config/gqrx/bookmarks.csv
Cannot connect to server socket err = File o directory non esistente
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock

RtApiAlsa::getDeviceInfo: snd_pcm_open error for device (hw:0,0), Dispositivo o risorsa occupata.

getDeviceList : Available input devices:
    0 : "HackRF HackRF One 4715cd"
    1 : "RFSPACE SDR-IQ Receiver"
    2 : "RFSPACE SDR-IP Receiver"
    3 : "RFSPACE NetSDR Receiver"
    4 : "RFSPACE Cloud-IQ Receiver"
    5 : "hw:HDA Intel PCH,0"
    6 : "default"
    7 : "HackRF One #0 644064dc214715cd"
    8 : "RTL-SDR Spectrum Server"
    9 : "Red Pitaya Transceiver Server"
    10 : "Complex Sampled (IQ) File"
Loading configuration from: "default.conf"
Configuration file: "/home/mauro/.config/gqrx/default.conf"
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.11
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 

FATAL: Wrong rtlsdr device index given.

Trying to fill up 1 missing channel(s) with null source(s).
This is being done to prevent the application from crashing
due to gnuradio bug #528.

********************
********************
No change in output device:
  old: 
  new: 

Failed to set RX input rate to 1.8e+06
Your device may not be working properly.

IQ DCR samp_rate: 1.8e+06
IQ DCR alpha: 5.55555e-07
Changing NB_RX quad rate: 96000 -> 1.8e+06
Requested sample rate: 1800000
Actual sample rate   : "1800000.000000"
New FFT rate: 25 Hz
Requested bandwidth: 1099999 Hz
Actual bandwidth   : 0 Hz
setFreqCorr : 0 ppm
New LNB LO: 0 Hz
updateHWFrequencyRange failed fetching new hardware frequency range
New mode index: 5
FM demod gain: 0.509296
Filter preset for mode 5 LO: -80000 HI: 80000
Generating taps for new filter   LO:-80000   HI:80000 TW:32000   Taps: 19
setFftRate to "25 fps"
New FFT rate: 25 Hz
New FFT rate: 25 Hz
setFftSize to "8192"
New FFT rate: 25 Hz
Generating taps for new filter   LO:-80000   HI:80000 TW:32000   Taps: 19
Configure I/O devices.
CIoConfig : Available output devices:
    0 : "Audio interno Stereo analogico"
New input device selected: 2
  Label: "HackRF One #0 644064dc214715cd"
  Devstr: "device='HackRF One',driver=hackrf,part_id=a000cb3c00604355,serial=0000000000000000644064dc214715cd,soapy=2,version=2018.01.1"
saveConfig
New LNB LO: 0 Hz
saveSettings *** FIXME_ SQL on/off
New FFT rate: 25 Hz
Loading configuration from: "/home/mauro/.config/gqrx/default.conf"
Configuration file: "/home/mauro/.config/gqrx/default.conf"
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.11
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
[INFO] Opening HackRF One #0 644064dc214715cd...

FATAL: setupStream invalid channel selection

Trying to fill up 1 missing channel(s) with null source(s).
This is being done to prevent the application from crashing
due to gnuradio bug #528.

********************
********************
No change in output device:
  old: 
  new: 

Failed to set RX input rate to 8e+06
Your device may not be working properly.

IQ DCR samp_rate: 8e+06
IQ DCR alpha: 1.25e-07
Changing WFM RX quad rate: 1.8e+06 -> 8e+06
Requested sample rate: 8000000
Actual sample rate   : "8000000.000000"
New FFT rate: 25 Hz
Requested bandwidth: 1099999 Hz
Actual bandwidth   : 0 Hz
setFreqCorr : 0 ppm
New LNB LO: 0 Hz
updateHWFrequencyRange failed fetching new hardware frequency range
New mode index: 5
Filter preset for mode 5 LO: -80000 HI: 80000
Generating taps for new filter   LO:-80000   HI:80000 TW:32000   Taps: 19
setFftRate to "25 fps"
New FFT rate: 25 Hz
New FFT rate: 25 Hz
setFftSize to "8192"
New FFT rate: 25 Hz
Generating taps for new filter   LO:-80000   HI:80000 TW:32000   Taps: 19
New FFT rate: 25 Hz
New FFT rate: 25 Hz
No audio FFT data.
No audio FFT data.
No audio FFT data.
No audio FFT data.
```

GravesJake, 2019-03-07:
Try selecting the "other" HackRF in the device list inside GQRX.

pelixm, 2019-03-08:
I did and everything worked fine, the "good" entry looks like the one with the shortest description.
Thanks for your help and regards

GravesJake, 2019-03-08:
No problem!

closed 2019-03-08 by GravesJake

zen1963, 2020-04-19:
Hi , 
can you explain me that : Try selecting the "other" HackRF in the device list inside GQRX. ? 
I don't understand at all !!

ChasJutaro, 2020-04-19:
Can u include a screenshot of the window containing the 'Other' option to clarify?

ChasJutaro, 2020-04-19:
When you use HackRF with GQRX, you will always have 2 options in the dropdown menu of choosing the device, in this case, it's listed as follows:
 _0 : "HackRF HackRF One 4715cd"
 7 : "HackRF One #0 644064dc214715cd"_ , hence, choose the one that you're not using right now, which would be "the other one".

zen1963, 2020-04-19:
i can include a screen shot , because I haven't  started yet to install hackrf one ! This page is the only on I found to install hackfr on ubuntu 18-04  ! That's why I want to understand this advice to follow it after following pelixm ... 
Thanks for you help ! I am a beginner !

zen1963, 2020-04-20:
Hi

zen1963, 2020-04-20:
> When you use HackRF with GQRX, you will always have 2 options in the dropdown menu of choosing the device, in this case, it's listed as follows:
> _0 : "HackRF HackRF One 4715cd" 7 : "HackRF One #0 [...]

Can you show me Where on the screen shot sent by pelixm , I Must select his option :Try selecting the "other" HackRF in the device list inside GQRX , to succeed  to install Hackrf ubuntu 18?
Thanks .-