greatscottgadgets/hackrf #610: Antenna analyzer with Relative function

Patnetpatnet, 2019-03-29:
Is anyone familiar with spectrum analyzer software that allows antenna return power calculations for VSWR. One needs to be able to zero out the power from the noise source so db level is flat. You can then add antennas to see reflected power and make SWR calculations. Pavels' Spektrum does this for RTL. Is there something similar for Hack RF One? Thanks.

GravesJake, 2019-05-24:
Sorry for the late response, I think a better place to ask something like this would be the hackrf irc channel or the mailing list.

miek, 2019-11-26:
qspectrumanalyzer does have this feature, but it's not super clear from the docs - here are some notes on how to use it: https://github.com/xmikos/qspectrumanalyzer/issues/70#issuecomment-557845699

closed 2019-11-26 by miek