greatscottgadgets/hackrf #611: Does 2Msps sampling rate affects Tx?

hopkinskong, 2019-04-01:
I understand there is a 8Msps (or 2Msps) minimum sampling rate requirements for the ADC (and baseband). Is it only applies to Rx only? How about Tx?

jboone, 2019-04-01:
@hopkinskong I don't see in the MAX5864 datasheet a minimum for the DAC. However, you will be limited by the width of the transmit baseband filter in the MAX2837. So at 2Msps, you will have alias images at multiples of the sampling frequency that aren't very well filtered. Additional filtering may be required, depending on your application.

hopkinskong, 2019-04-01:
Thanks for the info! As for the image frequencies, i suppose there is no way to do the filtering on software (i.e. GNURadio), right?

jboone, 2019-04-01:
@hopkinskong That's correct. At the point where these images are produced, you're no longer in the digital domain. So the solution would need to be analog in nature.

hopkinskong, 2019-04-01:
@hopkinskong Thanks! Where do this image frequency produced? Is it in the baseband IC? I have read that the IF frequency will also generate spurs in the image frequencies too.

jboone, 2019-04-01:
@hopkinskong It's produced from the act of converting digital samples into an analog voltage or power, and is due to sampling theory diverging from the actual behavior of electronics in the physical world. This takes place in the DAC portion of the MAX5864.

hopkinskong, 2019-04-01:
@jboone I see. Thanks again for the rapid response. So not only there are surplus image frequencies produced by the DAC in MAX5864, there are also the same thing happen in MAX2837 IF.

Closing the issue as it is now resolved.

closed 2019-04-01 by hopkinskong