greatscottgadgets/hackrf #612: Damage from reversed polarity

vojmar, 2019-04-20:
### Steps to reproduce
1. Accidently reverse dc input

### Actual behaviour
When connected over usb to pc 3v3 1v8 and RF diodes are lid up, but usb diode does not come on.

Some time ago my micro usb connector broke of, so i bought a replacement and forgot to check which side up does the cable go in. So i pluged it into my pc with great expectation and instead i blew something up. Some places of the board got quite hot. Now i am stuck in situation where i dont know what to replace to fix the problem. I dont know how to read gerber files and i am hoping i did not kill the microcontroller. I dont know if the damage comes from reversed input voltage or data lines (i am expecting those to be connected directly to controller). So far i had no luck solving the problem. Another hackrf is on its way so i can use it as an reference.

gpoleszuk, 2019-04-23:
Dear Michael Ossmann

related to this issue, what about to improve the hardware project (future version) including a "diode bridge" to reverse, naturally, the power input (-0.6V considering a silicon diode...)?

vojmar, 2019-04-29:
got my hands on another device,
Still not sure what to look for, seems like the voltage regulators are alright and so i have no idea where else is the dc input connected. I have borrowed cheap logic analyzer just for this purpose, but my poor understanding of gerber files prevents me from making any major breakthrough. There are some smd parts (probably rf chokes/filters) on gnd and +5v lines from usb, i wasnt able to find datasheet but everyone of them has different resistence and thats probably not a good sign.

miek, 2019-11-27:
I think it's going to be difficult to go through repairing that - there could be all sorts of parts that have failed.

Where did you purchase the HackRF? Given that the original fault was a bad USB connector, we would be willing to offer a replacement (assuming it's an official GSG unit).

closed 2019-12-16 by miek