greatscottgadgets/hackrf #614: Spikes and no signals

flanders-00, 2019-04-26:
Hello Everyone,

I have a problem with my new HackRF.

The situation is following:
When I start the HackRF by pressing play button in the SDRSharp (v1.0.0.1700 on Win10) , I can recieve  many of the FM station in frequency band ~ till 108MHz (see the screenshot) there are no problems. 
But if I turn the frequency up (for example to 170Mhz), then I'll see central static spike (it's normal, i know) and **many side spikes** that are moving towards each other while I turn up the frequency. Correct IQ make the central spike a bit less, but the side spikes does not sisappear.
But the worse thing is that within 140-200Mhz band I do not receive anything (no any modulated signals), only noise and DC spikes. Although I exactly know that there are many city services within this band and i'm receiving them by my rtl-sdr dongle.

I tried to use 5 different antennas and  I had the same result.
I tried to connect HackRF to different laptops (with different Windows) with the SDRSharp and I had the same result.
I repeatedly changed the firmware via Ubuntu and it did not help (now i have the latest 2018.01.1)

And one more issue, I tried to work via Gqrx (2.9) (via Ubuntu 18,.04,2), but when I tap the play button there is nothing happening. No curve of the signal, no waterfall, nothing,
although Gqrx works good with rtl-sdr dongle.

How do you thing, what it can be? Is it any hardware problem?

[image]

combatlord, 2019-04-26:
Do you have already updated the CPLD?
This will mostly forgotten and gives strange behaviors.

https://github.com/mossmann/hackrf/wiki/Updating-Firmware
See Step 2.

Can you show us the parameters for LNA-Gain, VGA-Gain and if you have enabled the AMP?
Dou you have the same results with a RTL-SDR Dongle?

flanders-00, 2019-04-26:
Yes i've updated CPLD, and did it one more time in accordance with the instruction (https://github.com/mossmann/hackrf/wiki/Updating-Firmware). The update was successfully, but there is no any improvement.

Please see Gains turning window. 
I've briefly enabled AMP a couple times, when i started to work with HackRF, but now it's off.
[image]

combatlord, 2019-04-27:
Do you see an amplifying to the noise floor when you enable the amp or adjust the gain?

The only thing that could be happened is that you maybe have a fail in the MGA-81563 amplifier (RX). So only very strong signals passes the Receiver.

Do you a radio amateur? This I ask because if you transmitt with a PA nearby the Hackrf you can destroy the input stage. -5dBm is the max with RX-AMP enabled. Have you connect a external LNA-Amplifer?

Do you receive nothing else on the band only FM-Broadcast?
Was the hackrf OK befor and Is this happened from now?

flanders-00, 2019-04-27:
**combatlord**, thank you for helping me.

Yes, the noise floor increases when I enable amp as well as I turn up the gains. 

Please see the FM band reaction to the gains increasing. I can hear clearly the FM stations even when VGA -8db and LNA - 16db (without amp of course).

I'm just beginner. I didn't  transmit anything nearby (my be only some one of my neighbors did?). No i haven't connected the external amplifier.     

Yes, that's right, only FM stations. No other signals except the spikes that are moving towards to each other and then goes to he opposite direction (converge and diverge) when I'm turning up the frequency. the spikes starts to "walk" by the band from approximately 115MHz.
Truly say I've got some peaks within 115-170MHz those seems like "normal" (not DC spikes) and those "tied" to the specific frequencies, but there are no any modulated signals on them. I think it's just some kind of noise.

I tried to receive from the different emplacements of the city and it's absolutely for sure that there are many signals from the city services withing the band till 180Mhz at those emplacements.

The HackRF is new and I'm just starting to work with it. So the problem since the beginning of use .
I used rtl-sdr before and it worked good.

[image]
[image]

combatlord, 2019-04-27:
This is really an odd behavior.
Like you have a FM-Broadcast bandfilter active.
But those one don't exists internal here. 
Only as an external filter.

To destroy the input stage you must realy close to the transmitter
or if the TX-Power is really big. But also 1 KW on HF with an
adequate antenna (yagi)  at a reasonable distance would'nt hurt the receiver. In this case you must sit before this 
And I don't think you sit on the neigbors roof ;-)

Which antenna do you use?
The SMA connector exists in 2 ways. SMA and SMA-RP.
On the first view it looks like the same but the SMA-RP ist reverse polarized. That means the Pin on the Male-Connector not exists.

The HackRf only uses SMA (not the RP-Variante).
The male-connector must have a pin inside.
Thats the only thing I know at the moment what could produce this behavior.

Do you know a frequency outside the broadcast band where you exactly know that are there strong signals?
In SDR-Sharp it's easy to switch the hardware so you can
easy compare if you see any little spikes on the Hackrf. 
You can also start two SDR-Sharp sessions and compare this at the same moment.

As I see on the picture that you have a bandwith of 10 Mhz.
When you zoom to it  (Slider on the right side) you will see a detailed spectrum. Do you see any signals which are modulatetd here?

regards Chris

virendera, 2019-04-27:
Are you did experiment with antenna while hackrf one is running and amp is on. Would you check it on strong broadcast fm station while both gain s are high and then turn on amp and then compare differences.

flanders-00, 2019-04-29:
**combatlord,**

I used 4 different antennas: VHF/UHF (2 pc., telescopic and non telescopic type), one for WIFI band (as on the router), and Nagoya NA-771. 
All of them have SMA connectors with the pin inside.

Using VHF/UHF antennas with RTL-SDR  i can receive the signals beyond the FM-band.  I found the strong modulated signal that often arise at the specific frequency. Then i ran the HackRF and moved to that frequency, and i saw absolutely nothing. No even little peak at that frequency, even when i used the zoom. 

**virendera,** 

Yes, i did experiments with antenna while HakRF was running, but truly say i don't remember was the AMP enabled at the moment or not. I enabled it just couple times when i started to work.
Please see the short video (attached) with reaction to the gains changing.
If I enable AMP (while gains are at the low levels), the noise floor is increasing by about 25dB. 

I don't know what to do with the device :)
Very strange situation.
May it be a partial damage of the LNA, or some chip, that not allows receive the signals at the frequencies upper the FM-broadcast band? 

[video1.zip](https://github.com/mossmann/hackrf/files/3128507/video1.zip)

virendera, 2019-04-30:
I see the video it's look like normal but you didn't turn on amp while all gain are at maximum level. You should do that to check differences, if all signals boosted then lna is not broken, if you see nothing while lna is on then lna is broken. You can check it on a strong broadcast fm station. Also remember that you always used your antenna connected to hackrf and all experiments with antenna should be do in offline mode. Even static charge can damage lna. You can check your tx lna with foober. You just need to install hackrf extension to broadcast and check it with rtl-sdr.

flanders-00, 2019-04-30:
Is that safety to enable AMP on a strong FM station while gains are at maximum? 
Doesn't it damage LNA?

virendera, 2019-04-30:
Just put your both gain on middle level and turn on amp.

flanders-00, 2019-04-30:
Thank you for the additional information.
Please see the attached short video with reaction to the AMP enabling. 
Is it normal response?

[Video2.zip](https://github.com/mossmann/hackrf/files/3131861/Video2.zip)

virendera, 2019-05-01:
It's look like your amp is not damage and working good, I think you need a better tune antenna. Is it a genuine hackrf or clone? Always remember for a specific frequency a right antenna is required. You say you are able to see some broadcast with your rtl-sdr but not hackrf. Always remember although hackrf and rtl-sdr both have equal 8bit adc but due complex rf path of hackrf, rtl-sdr  performance is good in tx mode comparison to hackrf. I personally observed it with my hackrf (clone) and rtl-sdr v3.

flanders-00, 2019-05-01:
Yes, you were right.

I turned both gains to maximum (without AMP) and then I saw some (not all) of the peaks, that i received  with rtl-sdr.
But the peaks were at the much more less level (in comparison with rtl-sdr).
Enabling amp increase overall noise floor but do not make the peak with the signals much better.

My HackRF is clone.
Is that the peculiarity of the HackRF related to it's low sensitivity? It's almost not possible to capture the signals with such low peaks level.

virendera, 2019-05-01:
Actually clone hackrf is not bad if manufacture not use cheap substitute parts. I think you should try with different antenna.

GravesJake, 2019-05-21:
Closing this as it seems to be resolved, feel free to re-open if you have more questions.
Thanks for all your help @combatlord and @virendera.

closed 2019-05-21 by GravesJake

pablozl85, 2023-03-27:
hello, I have a problem that my hackrf portapack h2 receives signals and does not send in any way reception tested on flipper zero car and other devices simply does not send a signal does anyone have any ideas for it? thanks in advance.