greatscottgadgets/hackrf #616: HackRF communicate with MCU board

bunphotk, 2019-05-10:
I am new in HackRF One. 
 
Can HackRF communicate with microcontroller board/mini PC? If yes, which board/HW is recommended for starter? 
 
I want to use HackRF with my TDOA project so I want also to hook GPS data into it. Are there any port on HackRF I can feed GPS timestamp?

Thanks

GravesJake, 2019-05-21:
There are various microcontroller boards that have RF TX/RX capabilities, but I'm a little confused about what you're looking for exactly. Could you elaborate?

bunphotk, 2019-05-22:
@GravesJake. I understand that HackRF can only send IQ data to computer via USB port (I might be wrong). What about use HackRF with a microcontroller board instead of PC? What is the best way to start with? Thanks.

GravesJake, 2019-05-23:
That's correct that the HackRF communicates with the host PC via USB. I'm not familiar with using other microcontroller boards with RF capabilities, a good place to ask about this would be the HackRF IRC channel or mailing list. Is it the USB communication itself that you're trying to circumvent?

virendera, 2019-08-27:
I think there is a microcontroller fabricated on  hackrf PCB also.

miek, 2019-11-26:
It's certainly possible to get the HackRF to communicate with another microcontroller, but it does not support it out of the box. It also doesn't support any sort of GPS timestamping currently, you would need to modify the firmware to implement that.

closed 2019-11-26 by miek