greatscottgadgets/hackrf #617: Can't upload the firmware to a new nxp lpc4320 chips

kangyuzhe666, 2019-05-14:
### Steps to reproduce
1.I bought a pcb and some chips through bom file then i constructed my own hackrf one.
2.I finished welding the pcb.
3.when i begin to upload the firmwire i meet some problems. 
4.first I make my broad enter the dfu mode.
5.when i use 'sudo make hackrf_usb-program' to upload the lpc its error 
kyz@kyz-Inspiron-7460:~/new/hackrf/firmware/hackrf_usb/build$ sudo make hackrf_usb-program
[ 14%] Built target libopencm3_hackrf_usb
[ 17%] Built target hackrf_usb_m0.elf
[ 17%] Built target hackrf_usb_m0.bin
[ 98%] Built target hackrf_usb_dfu_objects
[100%] Built target hackrf_usb_dfu.elf
[100%] Built target hackrf_usb_dfu.bin
dfu-suffix (dfu-util) 0.7

(C) 2011-2012 Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org

No valid DFU suffix signature
New DFU suffix added.
[100%] Built target hackrf_usb.dfu
dfu-util 0.7

Copyright 2005-2008 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2012 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org

Filter on vendor = 0x1fc9 product = 0x000c
Opening DFU capable USB device... ID 1fc9:000c
Run-time device DFU version 0100
Claiming USB DFU Runtime Interface...
Determining device status: state = dfuIDLE, status = 0
WARNING: Runtime device already in DFU state ?!?
Found Runtime: [1fc9:000c] devnum=0, cfg=1, intf=0, alt=0, name="DFU"
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0100
Device returned transfer size 2048
DFU CRC does not match
Warning: File has no DFU suffix
bytes_per_hash=690
Copying data from PC to DFU device
Starting download: [##################################################] finished!
unable to read DFU status
[100%] Built target hackrf_usb-program
kyz@kyz-Inspiron-7460:~/new/hackrf/firmware/hackrf_usb/build$ 

I use ubunut1804 dfu-util 0.7 .
I can upload the dfu file to the chip but when it loaded the broad will restart imediately and the flash upload failed.

GravesJake, 2019-05-21:
Just to clarify, you're able to load firmware in DFU mode but unable to use spiflash? Are you able to see the hackrf with `hackrf_info` after programming it in DFU mode?

kangyuzhe666, 2019-05-28:
> Just to clarify, you're able to load firmware in DFU mode but unable to use spiflash? Are you able to see the hackrf with `hackrf_info` after programming it in DFU mode?

Thinks to reply me. Because of my spiflash is empty ,I can't use `hackrf_info`.When I use dfu-util to upload the chip will restart and the program will disappeared. Forgive me for bad English I can't express clearly.

GravesJake, 2019-05-29:
So the board resets immediately after programming it in dfu mode?

kangyuzhe666, 2019-06-04:
> 因此，在dfu模式下编程后，电路板会立即复位吗？

Thank you very much for your answer, yes, it will restart after uploading ufu firmware.
I now suspect that the hardware problem is being ruled out. But I am very puzzled that I am completely following the standards and should not have such problems. I am troubled

miek, 2019-11-26:
This could be due to using a different SPI flash chip. I would try modifying the firmware to work around calls to `IAP_CMD_READ_SERIAL_NO`, and then try in DFU mode and see if the firmware runs OK. You may also try setting up JTAG and see if you get any HardFaults or hangs.

closed 2019-12-16 by miek