greatscottgadgets/hackrf #620: HackRF one is not working with compiled version of GNU Radio Companion

haronix-bg, 2019-06-10:
Hi,

### Version information
**Operating system**: Ubuntu 14.04.5 LTS

**hackrf_info output:** hackrf_info version: git-509c8f4
libhackrf version: git-509c8f4 (0.5)
Firmware Version: git-9bbbbbf (API:1.02)

GNU Radio Companion 3.7.10

This program is part of GNU Radio
GRC comes with ABSOLUTELY NO WARRANTY.
This is free software, and you are welcome to redistribute it.

### Output:
>>> Warning: This flow graph may not have flow control: no audio or RF hardware blocks found. Add a Misc->Throttle block to your flow graph to avoid CPU congestion.

Executing: /usr/bin/python2 -u /home/haronix-bg/SDR/RABOTESHTI_GRAFOVE/test.py

Osmocom Souce:
hackrf=0,buffers=32

When I try to compile hackrf via following steps:

cd hackrf/host
mkdir build
cd build
cmake ..
make
sudo make install
sudo ldconfig

gnuradio-3.7.10/gr-osmosdr/build$ sudo ldconfig
/sbin/ldconfig.real: "/usr/local/lib" is not a known library type

Any idea why the communciation between HackRF One & GNU Radio Companion (both versions are compiled from git reposity) is missing? No RF or hardware blocks found when all for osmocom blocks is available like blocks in GNU Radio Companion?

mossmann, 2019-06-12:
Are GNU Radio, gr-osmosdr, and libhackrf all installed under the same prefix (e.g. /usr/local/)?

haronix-bg, 2019-06-15:
That is the files located in /usr/local/bin$ ls

ccache-swig            gr_plot_psd            gr_time_raster_b
            gr_plot_psd_c          gr_time_raster_f
gnuradio-companion     gr_plot_psd_f          hackrf_clock
gnuradio-config-info   gr_plot_qt             hackrf_cpldjtag
grcc                   gr_plot_short          hackrf_debug
gr_constellation_plot  gr_psd_plot_b          hackrf_info
gr-ctrlport-monitor    gr_psd_plot_c          hackrf_operacake
gr-ctrlport-monitorc   gr_psd_plot_f          hackrf_spiflash
gr-ctrlport-monitoro   gr_psd_plot_i          hackrf_sweep
gr_filter_design       gr_psd_plot_s          hackrf_transfer
gr_modtool             gr_read_file_metadata  osmocom_fft
gr-perf-monitorx       gr_spectrogram_plot    osmocom_siggen
gr-perf-monitorxc      gr_spectrogram_plot_b  osmocom_siggen_nogui
gr-perf-monitorxo      gr_spectrogram_plot_c  osmocom_spectrum_sense
gr_plot_char           gr_spectrogram_plot_f  polar_channel_construction
gr_plot_const          gr_spectrogram_plot_i  swig
gr_plot_fft            gr_spectrogram_plot_s  usrp_flex
gr_plot_fft_c          gr_time_plot_b         usrp_flex_all
gr_plot_fft_f          gr_time_plot_c         usrp_flex_band
gr_plot_float          gr_time_plot_f         volk-config-info
gr_plot_int            gr_time_plot_i         volk_modtool
gr_plot_iq             gr_time_plot_s         volk_profile

haronix-bg, 2019-06-15:
One more messages:

Warning: restarting the docstring loader (crashed while loading 'osmosdr_sink')
Warning: restarting the docstring loader (crashed while loading 'osmosdr_source')
Warning: restarting the docstring loader (crashed while loading 'rtlsdr_source

haronix-bg, 2019-06-15:
You can see on the picture that there are unspecified blocks. The second one is crash of the gnuradio-companion after drag and drop of Osmocom Sink with the following message:

Warning: restarting the docstring loader (crashed while loading 'rtlsdr_source')
/usr/local/lib/python2.7/dist-packages/gnuradio/grc/main.py:54: Warning: Source ID 247 was not found when attempting to remove it
  gtk.main()
/usr/local/lib/python2.7/dist-packages/gnuradio/grc/main.py:54: Warning: Source ID 627 was not found when attempting to remove it
  gtk.main()
Segmentation fault (core dumped)

haronix-bg, 2019-06-15:
Picture

haronix-bg, 2019-06-15:
Picture

haronix-bg, 2019-06-15:
https://ibb.co/MPGv7N4

Hoernchen, 2019-06-16:
Just recompile & install gnuradio, then gr-osmosdr. Make sure not to mix different versions of gnuradio (package / your own version), and make sure to install gnuradio first, then build gr-osmosdr.

miek, 2019-11-26:
@haronix-bg sorry for the long delay, are you still having this issue with running GNURadio?

closed 2019-12-07 by miek