greatscottgadgets/hackrf #621: Receive a "Checksum Failure" and powers down during boot from USB stick

RexKlopfenstein, 2019-06-20:
### Steps to reproduce
1.
2.
3.

### Expected behaviour
Tell us what you expect should happen

### Actual behaviour
Tell us what happens instead

### Version information
**Operating system**:

**hackrf_info output:**

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
```
Insert any commandline or build output here
```

RexKlopfenstein, 2019-06-20:
I downloaded UnetBootin and executed with a formatted USB stick installed. In UnetBootin I selected ISO and pointed to Pentoo-full-amd64-hardened-latest.iso.  I restart with USB stick and proceeds until the checksums are calculated. The boot halts and requests a key stroke to power down.  The  problem file is syslinux.cfg.  I just purchased a hackRF and followed your install instructions when the above occurred.  If you can;t help, please point me in the right direction.

Rex K.

mossmann, 2019-06-21:
I suggest downloading the latest Ubuntu installation medium. You can use it to try Ubuntu without installing. In a terminal: `apt install gnuradio gr-osmosdr hackrf`.

closed 2019-11-26 by miek