greatscottgadgets/hackrf #622: hackrf_info  hackrf_info version: 2018.01.1 libhackrf version: 2018.01.1 (0.5) No HackRF boards found.

michealcorvin, 2019-06-21:
### Steps to reproduce
1. I upgraded to firware now I only see this 
#hackrf_info 
hackrf_info version: 2018.01.1
libhackrf version: 2018.01.1 (0.5)
No HackRF boards found.

2. I tried with the DFU bootloader but the of no use, the 3V3 green light and the TX Red light is always ON .The DFU bootlader does not even load or the lights flash it sits straight

3.How can it be fixed i want the device to run and before upgrading I was never able to tune into any fm station i used to get signals but i never hear any sound(language/songs) only loud noise ,i thought iw was the issue with the firmware ,hence upgraded it,

### Expected behaviour
I need the device to be working 

### Actual behaviour
Now it shows
# hackrf_info 
hackrf_info version: 2018.01.1
libhackrf version: 2018.01.1 (0.5)
No HackRF boards found.

I did a dmesg andi get:

 8614.702252] usb 3-2: device not accepting address 44, error -71
[ 8633.248918] usb 3-2: new high-speed USB device number 46 using xhci_hcd
[ 8633.389562] usb 3-2: New USB device found, idVendor=1fc9, idProduct=000c, bcdDevice= 1.00
[ 8633.389568] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 8633.389571] usb 3-2: Product: LPC
[ 8633.389573] usb 3-2: Manufacturer: NXP
[ 8633.389575] usb 3-2: SerialNumber: ABCD
[ 8677.652815] usb 3-2: USB disconnect, device number 46
[ 9445.987453] perf: interrupt took too long (6236 > 6203), lowering kernel.perf_event_max_sample_rate to 31800

### Version information
Parrot Os

**hackrf_info output:**

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
```
Insert any commandline or build output here
```

mossmann, 2019-06-21:
It looks like you do not have the firmware images (hackrf_one_usb.dfu and hackrf_one_usb.bin), so your firmware updates are failing.  You can find these files in the firmware-bin directory inside the latest release package: https://github.com/mossmann/hackrf/releases/tag/v2018.01.1

michealcorvin, 2019-06-22:
$hackrf_spiflash -w hackrf_one_usb.bin
File size 22024 bytes.
hackrf_open() failed: HackRF not found (-5)

hackrf_spiflash -w hackrf_one_usb.bin
File size 22024 bytes.
hackrf_open() failed: HackRF not found (-5)
[image]
[image]

michealcorvin, 2019-06-22:
Now i get this mgs after i  

LIBUSB_DEBUG=3 hackrf_info
hackrf_info version: 2018.01.1
libhackrf version: 2018.01.1 (0.5)
No HackRF boards found.

#dfu-util --device 1fc9:000c --alt 0 --download hackrf_one_usb.dfu
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: DFU suffix CRC does not match
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
dfu-util: No DFU capable USB device available

michealcorvin, 2019-06-22:
sudo hackrf_spiflash -w hackrf_jawbreaker_usb.bin
[sudo] password for india: 
File size 21176 bytes.
hackrf_open() failed: HackRF not found (-5)

$hackrf_info 
hackrf_info version: 2018.01.1
libhackrf version: 2018.01.1 (0.5)
No HackRF boards found.

miek, 2019-11-26:
Sorry for the long delay in getting back to you - I'm just working through the backlog of issues. Are you still unable to connect/program your HackRF?

febs, 2019-11-27:
Dropping by to say that I'm so glad that this project is feeling alive again. Thanks @miek

michealcorvin, 2019-11-29:
Hi thanks for the reply, But i m sorry to say i still face the same issue, i have kept the hardware aside now, cos i did not find any other solution till date from any1

miek, 2019-12-02:
Looking back on the information you posted before, it does suggest that the HackRF was launching the DFU bootloader & and doing a DFU flash successfully.

Would you mind giving it another go to launch in DFU mode and re-flash it? (I'm also on IRC for real-time support if that would help)

The full procedure is as follows:

- Hold the DFU button while plugging in the HackRF
You should see something like this in `dmesg`:

        [ 8100.751039] usb 1-1: new high-speed USB device number 16 using xhci_hcd
        [ 8100.900127] usb 1-1: New USB device found, idVendor=1fc9, idProduct=000c
        [ 8100.900132] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
        [ 8100.900135] usb 1-1: Product: LPC
        [ 8100.900138] usb 1-1: Manufacturer: NXP
        [ 8100.900141] usb 1-1: SerialNumber: ABCD

You should also see a device like this in `lsusb`:

        Bus 001 Device 017: ID 1fc9:000c NXP Semiconductors 

- Now you can run `dfu-util --device 1fc9:000c --alt 0 --download hackrf_one_usb.dfu`
You should see something like this in `dmesg`:

        [ 8554.534250] usb 1-1: USB disconnect, device number 17
        [ 8554.839619] usb 1-1: new high-speed USB device number 18 using xhci_hcd
        [ 8554.988618] usb 1-1: New USB device found, idVendor=1d50, idProduct=6089
        [ 8554.988623] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=4
        [ 8554.988626] usb 1-1: Product: HackRF One
        [ 8554.988629] usb 1-1: Manufacturer: Great Scott Gadgets
        [ 8554.988632] usb 1-1: SerialNumber: 0000000000000000457a63c92644461f

and something like this in `lsusb`:

        Bus 001 Device 018: ID 1d50:6089 OpenMoko, Inc.

The DFU flash is only temporary, so the above will need to be repeated if the HackRF is reset at this point.

- `hackrf_info` should display the device information. If it doesn't, try running the command as root
- You can now run `hackrf_spiflash -w hackrf_jawbreaker_usb.bin` to permanently flash the HackRF.

Please give that a go and let me know if any steps go wrong or do anything weird, and we can narrow down on any problems. Cheers!

michealcorvin, 2019-12-02:
[image]
[image]
[image]
[image]

miek, 2019-12-02:
Ah, you'll need to download the latest release from https://github.com/mossmann/hackrf/releases/tag/v2018.01.1, extract it, then either change directory into the `firmware-bin/` directory, or adjust the dfu-util command to point at the extracted file.

ie: `dfu-util --device 1fc9:000c --alt 0 --download path/to/firmware-bin/hackrf_one_usb.dfu`

michealcorvin, 2019-12-02:
Guys it worked THANKS A LOT!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

closed 2019-12-02 by michealcorvin