greatscottgadgets/hackrf #624: Issues getting repeat data transfers

leightonvillanueva, 2019-06-26:
### Steps to reproduce
1. Plug in HackRF.
2. Use signal generator to create a 50.5 MHz signal with -10 dBm.
3. Run my python script that calls hackrf_transfer 3 times on command line with the same settings except using different file names with a 5 second delay between calls.
4. Use Format-Hex command to print each files contents to console.

### Expected behaviour
Tell us what you expect should happen
Should print three transfer captures of the provided signal.
### Actual behaviour
Tell us what happens instead
First capture is successful providing useful data, next two captures are blank.
### Version information
**Operating system**:
Windows 10
**hackrf_info output:**
hackrf_info version: git-509c8f4
libhackrf version: git-509c8f4 (0.5)
Found HackRF
Index: 0
Serial number: 000000000000000015d463dc38a18c25
Board ID Number: 2 (HackRF One)
Firmware Version: 2017.02.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x006a434c

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
```
Insert any commandline or build output here
```
[image]
[image]
[image]
[image]

leightonvillanueva, 2019-06-26:
Here's my script:

`#!/usr/bin/env python2.7

from os import path
import sys
import subprocess
import time

if len(sys.argv) < 2:
    exit("No file name provided: Please add a file name to read from.")

file_name = sys.argv[1]

if path.exists(file_name):
    print("Executing " + file_name)
    f = open(file_name, "r")
    for line in f:
        subprocess.call(line)
        time.sleep(5)
    f.close()
else:
    print("Please create a " + file_name + " file!") `

And my commands.txt
[commands.txt](https://github.com/mossmann/hackrf/files/3332138/commands.txt)

GravesJake, 2019-06-27:
Can you check to see whether or not the RX LED turns on when expected?

leightonvillanueva, 2019-06-28:
@GravesJake  Yes, when using `hackrf_transfer -r test -f 50500000 -n 1024 -s 2400000 -a 1` the RX light illuminates and then turns off once the transfer is complete.

GravesJake, 2019-06-28:
Does this occur when using your script?

leightonvillanueva, 2019-06-28:
Yes it does.

GravesJake, 2019-06-28:
The light comes on separately for all three transmissions? Have you tried manually running the command three separate times and comparing the results?

leightonvillanueva, 2019-07-01:
I have tried manually running the command three separate times and the results are the same. First transfer produces results and the next two have no data.

GravesJake, 2019-07-03:
Sounds like a power saving issue. Try opening the hackrf in device manager and unchecking the box that says "Allow the computer to turn off this device to save power" under the Power Management tab.

leightonvillanueva, 2019-07-04:
I followed your steps and unchecked the boxes. Still having the same issues.

GravesJake, 2019-07-08:
Strange, is there anyway you can try it on a different machine or in a VM running a different OS?

closed 2019-11-27 by miek