greatscottgadgets/hackrf #625: Changing the Frequency to 5.9 GHz

rzn-290, 2019-07-02:
How can I change the frequency to 5.9 GHz? I followed the tutorial in Scott Gadgets and built a FM radio signal receiver (I think the FM radio uses 2.4 GHz). Now I need to change it #to 5.9 GHz for a DSRC project. Can you please help me? (I am a newbie so please don't mind if it is a basic question)

GravesJake, 2019-07-03:
The frequency in lesson 1 is about 96 MHz. If you're trying to use the existing flow graph you should be able to just change the channel_freq and center_freq variables to 5.9e9, and modify the WX GUI slider that suits your needs. You'll probably also need to play around with a few other settings to see what it is you're looking for.

Though, if all you're trying to do is get a visual for 5.9 GHz, I'd recommend using osmocom: `osmocom_fft -F -f 5900000000`

rzn-290, 2019-07-06:
Thank you so much! It worked.

closed 2019-07-06 by rzn-290