greatscottgadgets/hackrf #626: HackRF One does not transmitt all IoT frames

HalfDeadPie, 2019-07-02:
### Steps to reproduce
1. Install scapy-radio: https://github.com/BastilleResearch/scapy-radio
2. Try to send Z-Wave frame every 2 seconds
3. Watch the spectrum

### Expected behaviour
Example: sending Z-Wave frame every 2 seconds

### Actual behaviour
Not all Z-Wave frames are transmitted. I use different SDR to watch the spectrum and I can clearly see several frames are not transmitted. It looks like longer gaps between several frames:
|| ||| ||| || ||| |||
So every 3-4th frame is not transmitted

### Version information
**Operating system**: Linux Ubuntu 16.04 LTS

**hackrf_info output:** (I use RTL-SDR for reception and HackRF for transmission)
I have to respect formula on page 27 (https://apps.dtic.mil/dtic/tr/fulltext/u2/1054454.pdf).
HackRF transfer is able to transmit pre-modulated frames reliable. However, when I use GRC blocks, not all frames are transmitted. I don't know the reason and I don't understand how to debug that. When I increase interpolation of rational resampler from 250 to 500 or 750, all frames are transmitted, but they are not legitimate and Z-Wave devices do not receive them.

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

Gnuradio version: 3.7.9

### Output
```
$python top_block.py 
linux; GNU C++ version 5.3.1 20151219; Boost_105800; UHD_003.009.002-0-unknown

gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.9
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy redpitaya 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
Using Volk machine: avx_64_mmx_orc
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.9
built-in sink types: uhd hackrf bladerf redpitaya file 
Number of USB devices: 10
USB device 1d50:6089: 0000000000000000325866e622883323 match
Using HackRF One with firmware 2018.01.1 
UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU.....
```

GravesJake, 2019-07-03:
GNURadio is reporting lots of underruns (the appearance of "UUUUUUU.." at the end of the output). Your computer isn't able to provide the relevant frames fast enough. You can try changing your sample rates to fix this, but a better place to ask this would be the mailing list since it's not specific to HackRF failing.

GravesJake, 2019-07-03:
Actually after looking more closely at your screenshot, looks like gnuradio's trying to read from that socket and not getting enough to generate frames for the HackRF; so it's reporting underrun. So there's some sort of bottleneck before it actually gets to the HackRF.

HalfDeadPie, 2019-08-12:
Does GNURadio report underun according to configuration of blocks or always, when frames are not transmitted continuously, but with gaps?

Thank you for your previous answers, It explained me a lot.

miek, 2019-12-09:
The underrun reporting is an informal convention implemented by some blocks, it's not usually configurable and just prints out whenever the block doesn't have enough data to keep up with timing. There's an example here: https://github.com/osmocom/gr-osmosdr/blob/master/lib/hackrf/hackrf_sink_c.cc#L309

closed 2019-12-09 by miek