greatscottgadgets/hackrf #627: hackrf_sweep piping via awk then sort prints nothing

luoGanda, 2019-07-07:
This may be, or not be an issue of hackrf_sweep, but here it is.
Os is linux kubuntu.
Maybe this has to do with buffering or something.

This are commands being piped through a custom script hrfReport which has something like:
hackrf_sweep -f 1:300 2>/dev/null | awk -F "," "$awkScript"
Now, this are piping results
./hrfReport 1 300  <=ok, as expected
./hrfReport 1 300 | cat <=ok, repeatedly reporting to stdout
./hrfReport 1 300 | sort <=no output, something blocks
./hrfReport 1 300 | head -n4|sort <=ok, but prints just first 4 and then pipe is closed

I guess 'head' closes file descriptor or something and then 'sort' can read from std.

Can someone enlighten me with this?
Also using stdbuf -oL before commands doesn't help.
And in awk script there is fflush(); after circa 8 prints.

GravesJake, 2019-07-12:
I'm not super familiar with these file operations, but I'm guessing you're right in that there is some weirdness between the sweep output and what you're trying to do with it. Feel free to poke around at the code and see what you find.

miek, 2019-11-27:
The problem is that `sort` needs a finite input. Since `hackrf_sweep` never finishes, `sort` can't output anything - how would it know whether the next line should've been printed before the current output? To do something like this, you'd have to decide on a point where `hackrf_sweep` can end, then the sort can run.

closed 2019-11-27 by miek

luoGanda, 2019-11-28:
i thought that a finite input is an end of line - i think this is how it should work. Not sure - but i think this is how it works with for example rtlsdr.
I also tried with with fflush like mentioned before - but no luck.

miek, 2019-11-28:
`sort` only sorts whole lines in a file, so it needs to see the end of the file before it can sort

luoGanda, 2019-11-28:
yes - i though it's something like this.
With awk tool, it's meant to use sort every 1sec, so some workaround with EOF thing would probably work. I think that this works with rtlsdr directly.