greatscottgadgets/hackrf #629: Some clarification on libHackrf hackrf_transfer struct needed

JanosGit, 2019-07-18:
I'm building an SDR device abstraction layer and currently I'm working on the hackrf implementation.

As the source documentation is a bit thin at some points I need some clarification on the handling of the hackrf_transfer struct that is passed to rx and tx callbacks.

- Data format: What sample format is expected for the buffer? I assume it should be complex 8 bit signed integer values with a value range -127 to 127 with real part / imaginary part interleaved but I can't find any specific information on that..?
- buffer_length, valid_length: Whats do both mean? Do they count bytes or complex valued samples (that are 2 bytes per sample with real and imaginary part)? Am I allowed to change the value of one of those when transmitting to signal the device that I want to work on a smaller buffer size or do I always need to supply as many samples as requested?
- The API of my abstraction layer always invokes some "prepare to stream" callback prior to starting the stream which has a maximum buffer size the application should expect as an argument. This is helpful to do all non realtime safe buffer allocation work outside the rx/tx callback. However I found no way to query the HackRF how big the buffers used for rx/tx will be. Is there any way to get this information prior to the first callback or is there a buffer size that I can always expect?

Thank you for your help!

maltese, 2019-08-14:
I would also be interested in these answers!

straithe, 2021-11-12:
Thank you for raising this question, and sorry I haven't gotten to it sooner. I am working on these documentation updates. Unfortunately, it might take me a while as I am fairly booked. I will keep you updated!

mossmann, 2021-11-18:
Data format: Yes, the data in the buffer are signed 8-bit integer samples, interleaved I and Q.

buffer_length: The size in bytes of the data buffer for the USB transfer. This should be treated as read-only in a callback.

valid_length: The number of bytes in the buffer that were transferred to/from the USB device. This should be treated as read-only in a callback.

> Am I allowed to change the value of one of those when transmitting to signal the device that I want to work on a smaller buffer size or do I always need to supply as many samples as requested?

The buffer is a fixed size.

> I found no way to query the HackRF how big the buffers used for rx/tx will be. Is there any way to get this information prior to the first callback or is there a buffer size that I can always expect?

You can always expect the same buffer size (as specified in hackrf.c):
```
#define TRANSFER_BUFFER_SIZE 262144
```

PR #995 (2021-11-18): libhackrf: document hackrf_transfer struct
closes #629
closed 2021-11-18 by mossmann