greatscottgadgets/hackrf #631: error no file system?

TeleportingKitten, 2019-08-01:
I'm getting an error when I click the little red R icon when using the Capture, it's saying error at the top left and telling me I have no File system? I click OK and it brings me back to the Hack RF one capturing screen. It's definitely capturing because I see the spectrum, but am I missing files? I have a 64gb SD card inserted.

blardyblar, 2019-08-08:
Uh, what platform are you using it on? With what software? There isn't nearly enough information in your post to even begin to know what you're talking about... "little red R icon"?? "64gb sdcard"?? This means nothing without context!

miek, 2019-11-25:
This sounds like a Portapack issue. You probably just need to format the SD card with a FAT filesystem. For further info you should check out the Portapack repo/wiki: https://github.com/sharebrained/portapack-hackrf

closed 2019-11-25 by miek

Arefgh13, 2020-02-26:
> I'm getting an error when I click the little red R icon when using the Capture, it's saying error at the top left and telling me I have no File system? I click OK and it brings me back to the Hack [...]

How do you solve it, I have same problem

Arefgh13, 2020-02-26:
Please help I have no file system problem

miek, 2020-02-26:
See my reply above, you need to format the SD card correctly: https://github.com/sharebrained/portapack-hackrf/wiki/General-Use#microsd-card-socket