greatscottgadgets/hackrf #633: Can't see HackRF in Ubuntu 18.04

gdstorrick, 2019-08-04:
### Steps to reproduce
1. sudo apt-get install git build-essential cmake libusb-1.0-0-dev liblog4cpp5-dev libboost-dev libboost-system-dev libboost-thread-dev libboost-program-options-dev swig
2. git clone https://github.com/mossmann/hackrf.git
3. cd hackrf/host
4. mkdir build && cd build
5. cmake ../ -DINSTALL_UDEV_RULES=ON

### Expected behaviour
No make errors

### Actual behaviour
Returns with errors (see output).

### Version information
**Operating system**:
Clean (reset to factory) Ubuntu 18.04 on Dell XPS13
**hackrf_info output:**

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
```
-- Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE)
-- Could NOT find LIBUSB (missing: LIBUSB_INCLUDE_DIR)
-- Setting udev rule group to - plugdev
-- HackRF udev rules will be installed to '/etc/udev/rules.d' upon running 'make install'
CMake Error at /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
Could NOT find FFTW (missing: FFTW_LIBRARIES FFTW_INCLUDES)
Call Stack (most recent call first):
/usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
cmake/modules/FindFFTW.cmake:28 (find_package_handle_standard_args)
hackrf-tools/src/CMakeLists.txt:26 (find_package)

CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
LIBUSB_INCLUDE_DIR (ADVANCED)
used as include directory in directory /home/ghidora/hackrf/host/libhackrf
used as include directory in directory /home/ghidora/hackrf/host/libhackrf
used as include directory in directory /home/ghidora/hackrf/host/libhackrf
used as include directory in directory /home/ghidora/hackrf/host/libhackrf
used as include directory in directory /home/ghidora/hackrf/host/libhackrf
used as include directory in directory /home/ghidora/hackrf/host/libhackrf
used as include directory in directory /home/ghidora/hackrf/host/libhackrf/src
used as include directory in directory /home/ghidora/hackrf/host/libhackrf/src
used as include directory in directory /home/ghidora/hackrf/host/libhackrf/src
used as include directory in directory /home/ghidora/hackrf/host/libhackrf/src
used as include directory in directory /home/ghidora/hackrf/host/libhackrf/src
used as include directory in directory /home/ghidora/hackrf/host/libhackrf/src
used as include directory in directory /home/ghidora/hackrf/host/libhackrf/src
used as include directory in directory /home/ghidora/hackrf/host/libhackrf/src

-- Configuring incomplete, errors occurred!
See also "/home/ghidora/hackrf/host/build/CMakeFiles/CMakeOutput.log".
See also "/home/ghidora/hackrf/host/build/CMakeFiles/CMakeError.log".
```

gdstorrick, 2019-08-04:
More Information:

Did something change in LIBUSB that breaks the make?

I ran another test to see if I could blame this on Ubuntu 18.04. Perhaps not.

Exactly the same thing happens when I try the same steps in Kali running as root under VirtualBox on a Windows 7 machine:

root@kali:~/hackrf/host/build# cmake ../ -DINSTALL_UDEV_RULES=ON
-- The C compiler identification is GNU 8.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check if the system is big endian
-- Searching 16 bit integer
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of unsigned short
-- Check size of unsigned short - done
-- Using unsigned short
-- Check if the system is big endian - little endian
-- Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE) 
-- Could NOT find LIBUSB (missing: LIBUSB_INCLUDE_DIR) 
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Setting udev rule group to - plugdev
-- HackRF udev rules will be installed to '/etc/udev/rules.d' upon running 'make install'
CMake Error at /usr/share/cmake-3.13/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find FFTW (missing: FFTW_LIBRARIES FFTW_INCLUDES)
Call Stack (most recent call first):
  /usr/share/cmake-3.13/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  cmake/modules/FindFFTW.cmake:28 (find_package_handle_standard_args)
  hackrf-tools/src/CMakeLists.txt:26 (find_package)

CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
LIBUSB_INCLUDE_DIR (ADVANCED)
   used as include directory in directory /root/hackrf/host/libhackrf
   used as include directory in directory /root/hackrf/host/libhackrf
   used as include directory in directory /root/hackrf/host/libhackrf
   used as include directory in directory /root/hackrf/host/libhackrf
   used as include directory in directory /root/hackrf/host/libhackrf
   used as include directory in directory /root/hackrf/host/libhackrf
   used as include directory in directory /root/hackrf/host/libhackrf/src
   used as include directory in directory /root/hackrf/host/libhackrf/src
   used as include directory in directory /root/hackrf/host/libhackrf/src
   used as include directory in directory /root/hackrf/host/libhackrf/src
   used as include directory in directory /root/hackrf/host/libhackrf/src
   used as include directory in directory /root/hackrf/host/libhackrf/src
   used as include directory in directory /root/hackrf/host/libhackrf/src
   used as include directory in directory /root/hackrf/host/libhackrf/src

-- Configuring incomplete, errors occurred!
See also "/root/hackrf/host/build/CMakeFiles/CMakeOutput.log".
See also "/root/hackrf/host/build/CMakeFiles/CMakeError.log".
root@kali:~/hackrf/host/build#

Hoernchen, 2019-08-04:
No idea where your apt-get line comes from, just follow the actual readme from https://github.com/mossmann/hackrf/tree/master/host which includes pkgconfig which in turn is used to find libusb, and your build will work.

gdstorrick, 2019-08-04:
Thank you! Very well hidden.

closed 2019-08-04 by gdstorrick