greatscottgadgets/hackrf #634: HackRF_sweep dies after a while

blardyblar, 2019-08-08:
I'm using hackrf_sweep through a customized version of QSpectrumAnalyzer. I've noticed that hackrf_sweep stops working after some random number of times starting/stopping it. I have a feeling this is related to bug #481 

My use case is I tend to sweep for a few seconds, then stop, look at the data, then sweep again for another few seconds. This seems to aggravate the "infrequent" bug mentioned in #481. I'd say it crashes every 20 or 30 invocations of hackrf_sweep.

Using gqrx, the hackrf still works fine in non-sweep mode, but once sweep mode stops working, the only way to get it back is to press the reset button. I _believe_ that even unplugging / replugging the hackrf doesn't fix the issue, only actually pressing the reset button does, but I will verify and update the bug report when I get home tonight.

This makes it really annoying when it breaks, and I'm physically away from the hackrf.

Is it known roughly where this bug is? I'm quite comfortable diving into hackrf_sweep.c myself to fix it, but if there's already some progress / rough idea of where the bug is, no sense duplicating existing effort. In the other bug, something is mentioned about killing hackrf_sweep at the wrong time, and things not getting put back into an idle state.

Antanide, 2019-10-31:
What customized version of qspectrumanalyzer did you use? Thank you

blardyblar, 2020-08-07:
> What customized version of qspectrumanalyzer did you use? Thank you
It's my own customized version. I've put quite a bit of effort into making it more stable, especially hackrf sweep mode on windows, and adding other custom features for my application.

blardyblar, 2020-08-07:
This bug has been fixed now. It was causing kernel panics in the USB subsystem, but compiling the latest github firmware and hackrf_tools on 2020-08-04 the issue is now resolved.

closed 2020-08-07 by blardyblar