greatscottgadgets/hackrf #635: FATAL: Failed to open HackRF device (-5) HackRF not found

Thirumalll, 2019-08-19:
### Steps to reproduce
1.i'm woking on hackrf to transmit a jpg or video on particular frequency in television 
2.while i following this ( https://www.youtube.com/watch?v=z8DMFo4atnM )tutorial...i got the error which i mentioned in the screenshot 

### Version information
**Operating system**: UBUNTU

**hackrf_info output:**
https://www.youtube.com/watch?v=z8DMFo4atnM

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here. i'm using Gnu Radio Companion

[image]

miek, 2019-11-27:
Could you give a bit more detail on the problem? When the HackRF is plugged in, is it recognised by your OS (does it come up in `lsusb`)? If not: do any lights come on, and have you tried other USB cables?
If it is recognised, this may just be a permissions issue and there's some further detail here for fixing that: https://github.com/mossmann/hackrf/wiki/FAQ#i-cant-seem-to-access-my-hackrf-under-linux

closed 2019-12-07 by miek

pietrotedeschi, 2021-03-19:
I had the same issue and I discovered that it was due the USB cable.