greatscottgadgets/hackrf #637: RF gain not working, weakens signal

tomevans80, 2019-08-26:
### Steps to reproduce
1. In Gqrx (2.11.5), turn on RF amplifier gain

### Expected behaviour
Signal boosted.

### Actual behaviour
Signal significantly weakened by enabling it.

hackrf_info version: 2018.01.1
libhackrf version: 2018.01.1 (0.5)
Found HackRF
Index: 0
Serial number: 000000000000000057b068dc23997f63
Board ID Number: 2 (HackRF One)
Firmware Version: 2018.01.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x004b4746

I see this might be a common problem. I watched the tutorials first which was how I found out about the Hackrf, so I was very aware of always having an antenna connected during receive or transmit operations,but I have now read that changing the antenna could be a problem even when not in an RX/TX, this is the  only thing I might have done wrong I can think of.

virendera, 2019-08-27:
Issue #541 relate to this

miek, 2019-11-25:
This suggests that the RF amp has failed. If you get in contact with the reseller that sold you your HackRF One, they will replace it for you.

tomevans80, 2019-11-26:
> 
> 
> This suggests that the RF amp has failed. If you get in contact with the reseller that sold you your HackRF One, they will replace it for you.

Thanks for replying. I think this was the issue. The unit has been replaced so all sorted.

closed 2019-11-26 by tomevans80