greatscottgadgets/hackrf #638: Rx sensitivity of hackrf

virendera, 2019-08-27:
Hackrf is a deaf in comparison to RTL-sdr dongle and it's hard to find a weak signal on spectrum. Is there any way to increase it's sensitivity with some add on board?

jbaggs, 2019-09-03:
I've found the gain to be quite comparable. How are you setting AMP, LNA, and VGA?

miek, 2019-11-27:
One thing you can do to improve performance is to sample at the highest rate possible (20MHz), reduce the hardware low-pass filter bandwidth as low as possible (1.75MHz), and then decimate in software to increase SNR.

If you're still having trouble receiving weak signals, you may need to look at using an external LNA with filtering for your band of interest and possibly more suitable antennas with some gain, depending on the use-case.

closed 2019-11-27 by miek