greatscottgadgets/hackrf #640: firmware from master not initializing operacake

cavedon, 2019-09-06:
### Steps to reproduce
1. use hackrf + operacake
2. compile and load firmware from latest master (509c8f4)
3. execute hackrf_operacake -l

### Expected behaviour
```
Operacakes found: 
24
```

### Actual behaviour
```
Operacakes found: None
```

### Version information
**Operating system**:
Ubuntu Linux xenial amd64

**hackrf_info output:**
```
hackrf_info version: git-509c8f4
libhackrf version: git-509c8f4 (0.5)
Found HackRF
Index: 0
Serial number: 000000000000000087c867dc2912425f
Board ID Number: 2 (HackRF One)
Firmware Version: git-509c8f4 (API:1.03)
Part ID Number: 0xa000cb3c 0x00614f5e
Operacake found, address: 0x18
CPLD checksum: 0x8b150ff6
```

Looking at the firwmare code in `hackrf_usb.c`:
```
        hackrf_ui()->init();

        usb_run(&usb_device);

        rf_path_init(&rf_path);

        if( hackrf_ui() == NULL ) {
                operacake_init();
        }
```
It looks like `operacake_init()` can never be invoked, because if `hackrf_ui()` returned NULL, the line `hackrf_ui()->init()` would segfault.

Indeed commenting out the if around  `operacake_init()` worked for me in making the operacake board detected.

Maybe the comparison should be `hackrf_ui() == hackrf_ui_null`? However hackrf_ui_null is defined as static inside hackrf-ui.c.

Or is there a different firmware you recommend for use with operacake?

secsonthebeach, 2019-10-23:
I had the same problem. Was able to use git commit 72a03cf

miek, 2019-12-09:
I can reproduce this too.

The problem is that the firmware only calls `operacake_init` if `hackrf_ui()` returns `NULL` :

https://github.com/mossmann/hackrf/blob/master/firmware/hackrf_usb/hackrf_usb.c#L263

but `hackrf_ui()` never actually returns `NULL`, it still returns a valid pointer (to some stubbed out functions) even if no UI was detected.

jboone, 2019-12-10:
Ummm, good point... Oops.

It'd be nice to add a way to tell without having to add dependencies between the different UIs. Your idea of adding a function to the UI structure makes sense. I suppose we could get fancy and call it hackrf_ui_incompatibility() and return flags that indicate features that the active UI doesn't support. And then we can use those flags to gate at runtime what features we look for or try to enable? Regarding dependencies, that would require each UI to reference hackrf-ui.h for values of the flags. But that should be it -- and they already reference it for a declaration of the structure.

jboone, 2019-12-10:
...and regarding compatibility, I'm reminded of this, which might be relevant (GPIO vs. I2C mode for OperaCake): #575

PR #698 (2020-01-22): Operacake GPIO init fixes
This adds an extra function to the UI structure to specify whether the UI is compatible with the Operacake GPIO. This fixes #640 where Operacake GPIO mode would never be initialised. After applying this, `operacake -a 25 -g` (GPIO test) passes.

I also renamed the hackrf-ui files while I was there as they're the only source files to use dashes and it was annoying! :)
closed 2020-01-22 by mossmann