greatscottgadgets/hackrf #641: HackRF One Power Supply issue

jathro5, 2019-09-06:
I am having some problems with the HackRF one. As shown in the image, only the 3v3 and TX lights are on (the TX being very faint). It also can't be found by SDRsharp or GNU radio.

[image]

From what I found in the FAQ this means the power supply is shot. Is there a reset for this or a component that I could replace? 

The problem may have been caused by plugging it in to an USB/outlet adapter instead of a computer. We are using it for academic purposes at the University of Manitoba as a substitute for a Vector Network Analyzer while the other machines are in use. It is also possible that it was subject to a small DC current through the coaxial cable. 

### Version information
Using windows 10

d0ppelg4ng3r, 2019-10-06:
There are multiple power supplies. Do you have any electronic component repair experience? Access to a DMM (digital multi-meter) or O-Scope? I am an electronic repair technician and I would be willing to try and help you troubleshoot

miek, 2019-11-27:
This might not be a power supply issue. Could you try starting up the HackRF in DFU mode (by holding the DFU button while plugging in the USB cable), and see if it comes up OK as a USB device?

closed 2020-01-06 by miek

stoneburner, 2020-02-16:
I have the same issue, the hackrf worked fine before, i also tried to connect in dfu mode, no device was recognized.

i noticed that the lpc ic got warmer than usual.

i own several DMM and oscilloscopes, so i can measure at the testpoints, but i found no repair manual