greatscottgadgets/hackrf #644: Possible Tuning Issue?

fhhowdy, 2019-09-25:
### Steps to reproduce
1. HackRF with included antenna
2. gqrx, tune to frequencies below FM radio band

### Expected behaviour
I was expecting to see no FM radio signals.

### Actual behaviour
I'm seeing very strong WFM signals down to 77.9MHz.  At 80.9MHz, I'm getting 96.9 FM.  Is there some type of calibration process I need to follow?  It's strange that the frequencies are dead-on centered with what I'd expect from an actual fm radio (no weird offset)...but the frequencies extend well below normal broadcast range.

### Version information
**Operating system**: 
Mac OSX 10.15
HackRF One (just purchased from GS)
gqrx 2.5 software

**hackrf_info output:**

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.
gqrx 2.5 software
[image]

### Output
```
Insert any commandline or build output here
```

fhhowdy, 2019-09-25:
Actually, it goes to 73.9MHz  (89.9FM) ...  I haven't found any FM signals below this.  Both of these are 16MHz off from their true signal.  I can't imagine this is coincidental.  I'm thinking I have something configured incorrectly, but where, I don't know.  It works great, outside of the frequency deviation.
[image]

miek, 2019-11-25:
I've seen some issues like this with older firmware, or mismatched firmware/CPLD versions. Could you make sure you've got the latest CPLD and firmware images flashed? https://github.com/mossmann/hackrf/wiki/Updating-Firmware

closed 2019-12-16 by miek