greatscottgadgets/hackrf #649: Question about hackrf_transfer source code

Matioupi, 2019-10-15:
Hello,

i'm trying to understand how hackrf lib works and i am looking at hackrf transfer source code, and playing with it.
First thing I don't understand is why the frequency / amp are setted after the start_tx/start_rx are called ? I understand those actually work in a separate thread in some way, but I don't see/understand how those would wait for the frequency or other HackRF settigns to actually start.

Second point is the "sleep(1)"  on line 1062. If I remove this, then I got no more statistics in the while(hack_rf_is_streaming()) loop. It seems that the byte_count variable (which I understand is set in the tx/rx callback) seems to always be 0. It even seems i'm no more getting though the callback... and I'm not getting a clue why...

Matioupi, 2019-10-15:
I forgot to mention, I'm running the git master version & firmware, under an Ubuntu 19.04 x64 PC, hackrf hooked to USB 3 port (I only have USB3 on the laptop)

Matioupi, 2019-10-16:
I think I understand better what's happening for the sleep(1) removal :
when this is done, it happens that the execution enters the while loop before the tx callback actually gets called. 0 bytes have yet been transfered and then the last check :
```
if (byte_count_now == 0 && (hw_sync == false || hw_sync_enable == 0)) {
    exit_code = EXIT_FAILURE;
    fprintf(stderr, "\nCouldn't transfer any bytes for one second.\n");
    break;
}
```
			
exits the loop on a failure condition, then things get cleaned up and the callback is actually never called. I don't like this sleep(1) here. I would much better prefer something such as pthread_yield()
but many logics needs to be rewritten if the loop does not last at least 1s...

Matioupi, 2019-10-18:
line 873 of hackrf_transfer.c : `sample_rate_hz = (uint32_t)((double)sample_rate_hz * (1000000 - crystal_correct_ppm)/1000000+0.5);`
I don't understand where the +0.5 comes from...

miek, 2019-11-26:
> line 873 of hackrf_transfer.c : `sample_rate_hz = (uint32_t)((double)sample_rate_hz * (1000000 - crystal_correct_ppm)/1000000+0.5);`
> I don't understand where the +0.5 comes from...

The `+0.5` is so that it rounds to the nearest integer, rather than just rounding down with the `(uint32_t)` cast alone.

Matioupi, 2019-11-26:
with this hint, it now that looks obvious that this codes for a "nearest" function... thanks for the highlight !

closed 2019-11-26 by Matioupi