greatscottgadgets/hackrf #650: HackRF freezing after multiple calls to hackrf_start_tx/hackrf_stop_tx

Matioupi, 2019-10-16:
Hello,

I'm experiencing some random rate (but highly repeatable) freezes when trying to use hackrf though the hackrf lib API and especially hackrf_start_tx and hackrf_stop_tx.

When transmitting a large payload, but calling hackrf_start_tx only once, I have not been able to reproduce the freeze, but those occurs as soon as there are multiple calls.

I set up a minimal source code example that I hope will allow to reproduce the behaviour and tell me what I did wrong (most probable), or helps spotting a deeper burried issue :-)

The freeze seems to occur because hackrf_is_streaming() API function keeps returning HACKRF_TRUE despite the fact that it should have returned for a while.

### Steps to reproduce
1. Compile the minimal example in attachment : [hackrf_tx_freeze.zip](https://github.com/mossmann/hackrf/files/3733237/hackrf_tx_freeze.zip)
 `gcc hackrf_tx_freeze.c -lhackrf -o hackrf_tx_freeze`
2. Run the example :
 `./hackrf_tx_freeze`
3. Software will start broadcasting 1Mb "0" buffer on 433MHz (low gains...) in loop. 
Wait until it freeze with output such as :
`Just before calling hackrf_stop_tx
Just after calling hackrf_stop_tx
done TX round 187
Just before calling hackrf_start_tx
Just after calling hackrf_start_tx
    end of hackrf_is_streaming() wait loop after 0.065666s
    approx tx rate : 7984162.500000 Mech/s
Just before calling hackrf_stop_tx
Just after calling hackrf_stop_tx
done TX round 188
Just before calling hackrf_start_tx
Just after calling hackrf_start_tx
    stuck in hackrf_is_streaming() wait loop for 5.000000s, TX LED is probably still ON
    stuck in hackrf_is_streaming() wait loop for 5.000000s, TX LED is probably still ON`

### Expected behaviour
If I understand correctly, the software should not spend so many time (actually forever in the hackrf_is_streaming() waiting loop) and eventually exit normally after a while.

### Actual behaviour
At some point in time, which is very random, execution get stuck in the hackrf_is_streaming() waiting loop.
At this point, the TX LED is on and will not turn OFF without a manual reset.

I tried this on 2 different laptops (Ubuntu 19.04 and Ubuntu 18.04), one with USB2 and USB3 ports (as I saw some issues e.g. #500 that somehow looks similar but stated only USB3 as root cause) and with 2 different HackRF One units. Observed behaviour is the same.
I also saw closed #321 that match part of the observed behaviour (LED remains ON).
I have not investigated the behaviour on RX, but I also saw several issues #634, #624 that may have same origin ?

### Version information
**Operating system**: Ubuntu 18.04 and Ubuntu 19.04 on a Zbook 15G3 and a Shuttle DS68U7

**hackrf_info output:**

`hackrf_info version: git-e9c6c2d
libhackrf version: git-e9c6c2d (0.5)
Found HackRF
Index: 0
Serial number: 0000000000000000325866e6227e9423
Board ID Number: 2 (HackRF One)
Firmware Version: git-e9c6c2d (API:1.03)
Part ID Number: 0xa000cb3c 0x003e4f64
CPLD checksum: 0x05829db7`

and

`hackrf_info version: git-e9c6c2d
libhackrf version: git-e9c6c2d (0.5)
Found HackRF
Index: 0
Serial number: 0000000000000000325866e621880b23
Board ID Number: 2 (HackRF One)
Firmware Version: git-e9c6c2d (API:1.03)
Part ID Number: 0xa000cb3c 0x00564755
CPLD checksum: 0x05829db7`

Matioupi, 2019-10-16:
it seems like changing the hackrf.c stop_tx_function like this :

`int ADDCALL hackrf_stop_tx(hackrf_device* device)
{
	int result;
	result = kill_transfer_thread(device);
	if (result != HACKRF_SUCCESS)
	{
		return result;
	}

	return hackrf_set_transceiver_mode(device, HACKRF_TRANSCEIVER_MODE_OFF);
}`

solves my issue... and somehow, it make more sens to me first joining the thread before changing other stuff (if the original code, the receiver is first set to HACKRF_TRANSCEIVER_MODE_OFF which call some libusb stuff, and then the transfer thread is joined.

Matioupi, 2019-12-06:
This a just a bottle at sea to check if some official maintainers had time to check this issue and the PR i made for it.
I saw that a large cleanup in issues occurred recently so I thought this would end up in evolutions toward a new release and review of some PR.
Any news on the HackRF project in general ? Is it still alive ?
Regards to all GSG team !

miek, 2019-12-06:
I've not had time to look at this yet but there is a big push to resolve outstanding issues right now, and I'll be getting to this as soon as I can.

My plan over the next few weeks is to work on preparation & testing for a new release, so the project is definitely still alive!

closed 2021-02-06 by miek

Potomac, 2021-04-28:
This commit triggers crash with recent hackRF one chinese model, when using GQRX software, I am sure because I made a git bisect, see my bugreport : 
https://github.com/mossmann/hackrf/issues/883

That means that we have to find a better solution for solving the issue of @Matioupi, something that can not create problems with hackRF one chinese models.

ivan-kulikov-dev, 2021-05-04:
I have problem with hackRF one chinese model (bought in 2017 year) with hackrf-2021.03.1 firmware too.

Localacct21, 2021-07-19:
> I have problem with hackRF one chinese model (bought in 2017 year) with hackrf-2021.03.1 firmware too.

What is diffrent from chinese version except the rf front end? @cjheath