greatscottgadgets/hackrf #653: HackRF One No Rx/Tx

antman1p, 2019-10-18:
### Steps to reproduce
1. Run lesson1.grc from the HackRF One video tutorial on gnuradio
2. Also tried from windows SDR#

### Expected behaviour
I expect to see all lights on the HackRF one turn on and FM Radio signals in the FFT Plot.
### Actual behaviour
All lights on the HackRF One are on, but I see absolutely no signals in the FFT Plot
### Version information
**Operating system**: Pentoo

**hackrf_info output:** 
pentoo@pentoo ~ $ hackrf_info
hackrf_info version: 2018.01.1
libhackrf version: 2018.01.1 (0.5)
Found HackRF
Index: 0
Serial number: 000000000000000014d463dc2f7e6ce1
Board ID Number: 2 (HackRF One)
Firmware Version: 2018.01.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x006c475c

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

GNU Radio Companion 3.7.13.5
[image]

### Output
```
Insert any commandline or build output here
```

antman1p, 2019-10-18:
`pentoo@pentoo ~ $ hackrf_transfer -r -
call hackrf_set_sample_rate(10000000 Hz/10.000 MHz)
call hackrf_set_freq(900000000 Hz/900.000 MHz)
Stop with Ctrl-C
 0.0 MiB / 1.000 sec =  0.0 MiB/second

Couldn't transfer any bytes for one second.

Exiting... hackrf_is_streaming() result: HACKRF_TRUE (1)
Total time: 1.00025 s
hackrf_stop_rx() done
hackrf_close() done
hackrf_exit() done
fclose(fd) done
exit`

antman1p, 2019-11-14:
Following up...

mossmann, 2019-11-25:
Can you please try:

- repeating with a different USB cable

- `hackrf_cpldjtag -x firmware/cpld/sgpio_if/default.xsvf`

antman1p, 2019-11-25:
> Can you please try:
> 
> * repeating with a different USB cable
> * `hackrf_cpldjtag -x firmware/cpld/sgpio_if/default.xsvf`

```
pentoo@pentoo ~/Documents $ hackrf_cpldjtag -x firmware/cpld/sgpio_if/default.xsvf
Failed to open file: firmware/cpld/sgpio_if/default.xsvf
```

miek, 2019-11-25:
Ah, you can grab that file from the tar archive on the latest release: https://github.com/mossmann/hackrf/releases/tag/v2018.01.1

antman1p, 2019-11-25:
That worked!  I am now RXing.  Was that simply from using a different USB cord, or by updating the CPLD?  I feel that I did that step when I updated the firmware, weeks ago.

miek, 2019-11-25:
Excellent! It could've been either. You could try the previous USB cable again and see if the problem comes back, if you want to be sure.

antman1p, 2019-11-25:
For sure.  Thank you!

closed 2019-11-25 by miek

tekeke12, 2022-11-14:
> Ah, you can grab that file from the tar archive on the latest release: https://github.com/mossmann/hackrf/releases/tag/v2018.01.1

please tell me how to install this tar file ?

martinling, 2022-11-14:
@tekeke12 this issue is over three years old and refers to outdated versions of the HackRF firmware. Whatever problem you're having, that file is not the answer. Please open a new issue.

tekeke12, 2022-11-18:
C:\Program Files\GNURadio-3.7\bin>hackrf_transfer -t -
call hackrf_set_sample_rate(10000000 Hz/10.000 MHz)
call hackrf_set_freq(900000000 Hz/900.000 MHz)
Stop with Ctrl-C
 0.3 MiB / 1.016 sec =  0.3 MiB/second
 0.0 MiB / 1.016 sec =  0.0 MiB/second

Couldn't transfer any bytes for one second.

Exiting... hackrf_is_streaming() result: HACKRF_TRUE (1)
Total time: 2.03118 s
Caught signal 0
Caught signal 0
Caught signal 0
hackrf_stop_tx() done
Caught signal 0
hackrf_close() done
hackrf_exit() done
fclose(fd) done
exit

i can't tx

tekeke12, 2022-11-18:
C:\Program Files\GNURadio-3.7\bin>hackrf_info
hackrf_info version: git-cc7f599
libhackrf version: git-cc7f599 (0.5)
Found HackRF
Index: 0
Serial number: 0000000000000000f77c60dc26706bc3
Board ID Number: 2 (HackRF One)
Firmware Version: 2017.02.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x00584f58