greatscottgadgets/hackrf #655: HackRF Radio FM Signals

GiacomoNasi, 2019-10-21:
Hi everyone, I'd like to know if is there any way to get the FM audio from a specific frequency coming from an HackRF device.
I'd have to use this feature in my software, so I need a command line tool to call or a Java library.

Thanks for your help :)

GravesJake, 2019-10-21:
I'm not exactly sure what you're after, are you trying to listen to the FM audio or do something else with it, ie. save it to file?

GiacomoNasi, 2019-10-22:
No, I have to reproduce the audio in real time. If there's no way to do it, also saving the audio file and listening to it in a second moment could be ok :)

febs, 2019-11-02:
Did you watch the videos from Ossmann? The first one does exactly that in gnu radio.

miek, 2019-11-26:
As @febs says, have a look at the Software Defined Radio with HackRF video series here: https://greatscottgadgets.com/sdr/ - lesson 1 has a demo of receiving broadcast FM radio.

closed 2019-11-26 by miek