greatscottgadgets/hackrf #657: hackrf_transfer -r only one Time, until USB reconnect

chklein, 2019-10-22:
### Steps to reproduce
1. 1st call hackrf_transfer -r /dev/null
    Everything seems to be ok
2. 2nd call hackrf_transfer -r /dev/null
    I get the following issue:
    root@kali:~# hackrf_transfer -r /dev/null
call hackrf_set_sample_rate(10000000 Hz/10.000 MHz)
call hackrf_set_hw_sync_mode(0)
call hackrf_set_freq(900000000 Hz/900.000 MHz)
Stop with Ctrl-C
 0.0 MiB / 1.001 sec =  0.0 MiB/second

Couldn't transfer any bytes for one second.

Exiting... hackrf_is_streaming() result: HACKRF_TRUE (1)
Total time: 1.00146 s
hackrf_stop_rx() done
hackrf_close() done
hackrf_exit() done
fclose(fd) done
exit

3. After a USB reconnect it works one time again

### Expected behaviour
Work more than one time

### Actual behaviour
Doesn't work after the 1st try

### Version information
Kali 2019.3

**hackrf_info output:**

root@kali:~# hackrf_info 
hackrf_info version: git-e9c6c2d
libhackrf version: unknown (0.5)
Found HackRF
Index: 0
Serial number: 0000000000000000a74461c83856679f
Board ID Number: 2 (HackRF One)
Firmware Version: 2018.01.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x005f4f6a
hackrf_info: symbol lookup error: hackrf_info: undefined symbol: hackrf_cpld_checksum

Insert any commandline or build output here

Matioupi, 2019-10-23:
I don't know if this is the root cause, but your hackrf host tools and hackrf firmware are not synced :
hackrf firmware is 2018.01.1 and you are using hackrf tools git-e9c6c2d
you should try reverting your tools to the release version 2018.01.1, or update your firmware and clpd (by the way, you are receiving a message about not being able to retrieve the clpd checksum because i think you need a later fw version with usb APU 0x01:0x03 instead of 0x01:0x02

chklein, 2019-10-25:
Thanks for your Reply.

I builded and installed the hackrf host tools new from Release 2018.01.1

hackrf_info output is now:

hackrf_info version: 2018.01.1
libhackrf version: unknown (0.5)
Found HackRF
Index: 0
Serial number: 0000000000000000a74461c83856679f
Board ID Number: 2 (HackRF One)
Firmware Version: 2018.01.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x005f4f6a

The problem is the same like before. Do i have to do something with libhackrf? how can i build and install it?

When i upgraded to Firmware 2018.01.1, i upgraded the clpd too. So i think that is not the problem.

Many thanks

chklein, 2019-10-25:
I had to remove the Package libhackrf0 with apt-get remove libhackrf0.

Then i did the "make install" again.

Now the output from hackrf_info is:

hackrf_info version: 2018.01.1
libhackrf version: 2018.01.1 (0.5)
Found HackRF
Index: 0
Serial number: 0000000000000000a74461c83856679f
Board ID Number: 2 (HackRF One)
Firmware Version: 2018.01.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x005f4f6a

So i think now it looks fine, but the problem i can do only one time "hackrf_transfer -r /dev/null" further exists.

Thanks for any answer

closed 2021-02-25 by straithe