greatscottgadgets/hackrf #659: HackRF One for GPS-SDR

AgileEngineeringLLC, 2019-11-06:
I apologize if this has been discussed before. I'm interested in setting up a GPS-SDR using the HackRF One as the front end. Has anyone here tried this, and what SRD software did you use? Was it fairly intuitive to set up, or not so? What were the big roadblocks or challenges?

My interest is in setting up a GPS-SDR receiver, just a simple L1 configuration at first.

Thanks!!

FamerWaitingforRain, 2019-11-25:
There is a project named gnss-sdr,it's an open source GNSS software defined receiver.I think it support hackrf one.I also interest in the gps-sdr,we can do it together.

miek, 2019-11-26:
As suggested above, check out https://github.com/gnss-sdr/gnss-sdr

closed 2019-11-26 by miek