greatscottgadgets/hackrf #660: alignment of transmitted signal

Jakub6, 2019-11-07:
Hello, 

I am using the ADSB-Out library  (https://github.com/lyusupov/ADSB-Out) to generate an adsb message and transmit via hackrf. The procedure is to generate a message (4096 bytes) and alignment to 256K buffer size.

Instructions from the  ADBS library:
   - creating ADSB message and save to Samples.iq8s

   - Alignment to buffer size 256K
      $ dd if=Samples.iq8s of=Samples_256K.iq8s bs=4k seek=63
      1+0 records in
      1+0 records out
      4096 bytes (4.1 kB) copied, 0.00110421 s, 3.7 MB/s
      $

   -and then transmit via hackrf:
      hackrf_transfer -t Samples_256K.iq8s -f 1090000000 -s 2000000 -x 0

I suppose that this procedure just take the message and the rest of the 256K buffer fill with zeros and transmit it, so I would assume that I should receive just the message, but I received tens of the same messages.

Any idea why this is happening? Is there any problem with the buffer size? Could be the signal retransmitted so many times?

Thanks,

Jakub

Matioupi, 2019-11-08:
The dd command is padding the useful signal with "0x00" bytes, so it will not change the number of times it's transmitted.

I guess this would allow the 1090MHz carrier to start earlier and may helps it being detected at receiver side. Still you can test that it's not necessary (at least on my tests, I did not find it usefull).

Regarding the adsb out rules by themselves, the standard states that you should not transmit more than 6.2 msg/s average (from ICAO 10.IV 3.1.2.8.9)

For you tests, unless you are in a controlled environment, I advise you keep brodcasting / detecting on 868 MHz rather than using the actual 1090MHz reserved frequency.

miek, 2019-12-08:
I've reproduced this just now, following the instructions you gave.

I see the signal start to repeat every 8.192ms:
[image]

Doing the maths based on a 2Msps TX rate:

    $ units "2MHz * 8.192ms"
        Definition: 16384

says that it's repeating every 16384 samples, which is the size of one USB bulk buffer in the HackRF firmware. I'm guessing that when `hackrf_transfer` stops sending samples, the firmware is just repeating the previous buffer until the 'stop TX' command comes through.

As a workaround, you can pad your file with some extra zeros on the end (with the same 256KiB alignment):

    $ cp Samples_256K.cs8 Samples_256K_padded.cs8
    $ dd if=/dev/zero bs=16384 count=16 >> Samples_256K_padded.cs8
    16+0 records in
    16+0 records out
    262144 bytes (262 kB, 256 KiB) copied, 0.000244969 s, 1.1 GB/s

For me, using that file with `hackrf_transfer` doesn't send extra repetitions anymore.

straithe, 2021-03-03:
I'm going to close this as there hasn't been a response in a while, but please re-open this issue or open a new one if you still need assistance.

closed 2021-03-03 by straithe

mikotaFull, 2022-09-24:
hello,
I tried what miek recommends and I still get the signal repeated 160 times. I use gr-air-modes to decode the signal and as you can see on the picture it is repeated 160 times.

Here is the sequence of commands I use:
ADSB_Encoder.py 0xABCDEF 12.34 56.78 9999.0
dd if=Samples.iq8s of=Samples_256K.iq8s bs=4k seek=63
cp Samples_256K.iq8s Samples_256K_padded.iq8s
dd if=/dev/zero bs=16384 count=16 >>samples_256K_padded.iq8s
hackrf_transfer -t Samples_256K_padded.iq8s -f 1090000000 -s 2000000 -x 10 -a 1

it is right?
moreover I notice that the file Samples_256K_padded.iq8s is repeated twice which does not seem normal to me.
please could you help me to solve this problem?
Thank you.

[image]