greatscottgadgets/hackrf #665: How to read the firmware documents???help me

FamerWaitingforRain, 2019-11-25:
### research the code of hackrf one's MCU(LPC4320)
I downloaded  all the files of hackrf one project. And i want to konw how the mcu control these chip work.So i read the firmware document,but there are so many files,and exist several main function.I donot konw where is the first of these codes.
let's discuss.
[image]

miek, 2019-11-25:
The main entry point for the standard HackRF One firmware is in `hackrf_usb/hackrf_usb.c`. Other `main` functions are for test/demo applications.

FamerWaitingforRain, 2019-11-25:
thanks so much

> The main entry point for the standard HackRF One firmware is in `hackrf_usb/hackrf_usb.c`. Other `main` functions are for test/demo applications.

thanks so much

closed 2019-11-25 by miek