greatscottgadgets/hackrf #666: No HackRF boards found. (not the cable)

herrmanns, 2019-11-25:
### Steps to reproduce
1. connected hackrf device
2. No HackRF boards found.
3. tried seveal different cables
4. removed havoc display
5. dfu-util --device 1fc9:000c --alt 0 --download /*no luck
6. hackrf_cpldjtag -x /*no luck

### Actual behaviour
no device found

### Version information
**Operating system**:
Linux cf19 5.3.0-kali2-amd64 #1 SMP Debian 5.3.9-3kali1 (2019-11-20) x86_64 GNU/Linux

**hackrf_info output:**

hackrf_info version: git-e9c6c2d
libhackrf version: git-e9c6c2d (0.5)
No HackRF boards found.

whatever i did this bloody hackrf device not answer anymore. never had this kinda issues in the past
have the havoc portapack stuff b1e72c78 running btw

thx for help in advance.

miek, 2019-11-25:
Please could you try running `dmesg -w` in a terminal, then plug in the HackRF and paste any output that appears after plugging it in. Thanks!

herrmanns, 2019-11-25:
thx miek for help. dmesg keeps quiet. nothing appears when i stick it in. not in dfu mode nor in "regular" operating mode.

miek, 2019-11-25:
Do any lights come on on the HackRF when it's plugged in? Have you tried connecting it to different USB ports on your PC, or tried connecting it to a different PC?

cardre, 2019-11-25:
Or tried different USB cables? (some are power only and no data lines connected)

herrmanns, 2019-11-25:
hackrf works like charm. i have the portapack installed and could operate without any issues. 
the only thing is i can't connect via usb as you can see.
i tried it with my windows pc as well but nothing either. i checked the cables with other devices and cables are OK too.
from the antenna side the red orange and green LEDs are on. 
really weird. device works but no usb.

miek, 2019-11-25:
Just to confirm: when you try DFU are you holding down the DFU button on the HackRF, then plugging in the USB to put it into DFU mode? And there's still nothing in `dmesg`?

herrmanns, 2019-11-25:
sry i am late. yes exactly nothing happens on dmesg output even with the button procedure.

miek, 2019-11-25:
This sounds like it must be a hardware problem then (most likely a broken USB connector). If you get in touch with the reseller that you bought your HackRF One from, they will replace it.

herrmanns, 2019-11-25:
hm yes that smells like it. i am located in germany and bought it in the US thats terrible but i guess there is no other chance than sending back to sharebrained. really odd this usb thingy. ve never had that with any device yet.
thx a lot for help!

jboone, 2019-11-25:
Does the device not respond over USB even when you enter "HackRF Mode" from the PortaPack menu?

The USB connector is on the HackRF, not the PortaPack. So if you return the HackRF, you would want to return it to whoever shipped you the HackRF.

herrmanns, 2019-11-25:
no response jboone even not in "hackrf mode"
i thought i bought it there but you are right it was only the portapack. hackrf board was somewhere else. have to check it (but i think it was from US  as well)

miek, 2019-11-26:
I'm going to close this now, but if you have any issues with getting a replacement from the reseller please do get back in touch!

miek, 2019-11-26:
Where did you purchase the HackRF from? We may be able to arrange to exchange the unit at either Black Hat Europe or at Chaos Communication Congress, if that would help.

closed 2019-12-16 by miek