greatscottgadgets/hackrf #668: hackrf does not turn on ic tps62410 gets very hot

uglurass, 2019-11-26:
### Steps to reproduce
1. will not turn on unless hackrf is connected to 1+amp powersupply 
2. ic tps62410 gets very hot 
3.hackrf died after i connected LNA with not approved 12volt supply 
hackrf will draw 250mah but not turn on but if i disconnect charger leaving micro usb connected 
and connect charger the hackrf comes to life and draws 1000mah i dont allow it to be on for more than a few seconds any idea ? 
### Expected behaviour
Tell us what you expect should happen

### Actual behaviour
Tell us what happens instead
s
### Version information
**Operating system**:
win10
**hackrf_info output:**

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.
s
### Output
```
Insert any commandline or build output here
```c

miek, 2019-11-26:
When you connected the LNA, was that putting 12V on the HackRF antenna port?

uglurass, 2019-11-26:
yeah i think so.

miek, 2019-11-27:
I suspect something has probably failed short in the RF frontend, and that is then drawing a lot of current on one of the supplies causing the TPS62410 to heat up. I don't think there's any simple solution, it would need a bunch of troubleshooting & some good SMD soldering skills to replace any failed parts.

closed 2019-11-29 by uglurass

aripa2005, 2020-05-10:
How do you resolve the situation ? a have the same problem after a conect antena to CB station antena in car, on a fast charger. please let me know

miek, 2021-05-10:
Potentially anything on the VAA/VCC rails could've been damaged as 12V on the antenna connector would put up to ~11V on the VAA rail through the body diode on Q4.
I would try isolating the VAA rail by either removing Q3 or cutting the NT3 jumper trace. If it works without getting hot, then the fault is on the VAA rail and you could try powering the VAA rail from a current-limited power supply and see if any part gets hot.

uglurass, 2021-05-16:
> Yesterday I did the same mistake with the same result. The 62410 is
> getting hot quickly and after about a minute the LED's start flickering and
> there is a ticking sound from the inductors around the 62410. As I
> understand it, the most suspected DC path is via VAA, Q4, L134 to a half
> shorted D1. I removed D1 to cut that path but that did not help. Also
> connected the hackrf to a LiIon USB battery in order not to wake up the USB
> connection, but it is still getting hot. However, before the 62419 starts
> ticking, I can run hackrf_info and get sane results, so the microcontroller
> seems to be alive, at least for a minute or so.
> Any hints on further debugging or other ideas are appreciated.

it neets resistor

bernerus, 2021-05-17:
I think a physical jumper would make the product more safe against different environments,
Then the user could be specifically informed of the dangers of enabling the integrated bias tee circuit.

In my case, I have a mast top box with a preamp and a coax relay needing a lot more voltage than
the tiny 3.2V that Hackrf can provide, thus there is an external bias tee providing 12V up to the coax relay and the preamp
on receive. On transmit, the coax relay shifts the antenna to a separate transmit coax, but that's another story.

bernerus, 2021-05-18:
Well, since I accepted the fact that I probably killed my HackRF, I decided to dare doing some
experiments and troubleshooting in order to pinpoint what has broken.

First I cut all of the NTx traces, thus removing all load from the power management circuits.
Then I could measure the current flowing into VCC (and VAA). Turned out to be around 220 mA
which I found reasonable. 

However, even with all NTx traces cut, the power regulator still got hot, so I measured the resistance to
ground at the regulator end of the now cut NT2.Turned out to be 14.5 ohms, both ways. Which absolutely
explains why the regulator is getting hot. 

Then I conducted another experiment to see what is working except for the regulator. 
If Q5 is turned off  by connecting a 2K2 resistor from VBUS to VBUSCTRL, no voltage is presented to the regulator
even in the USB is feeding in 5V. 
Next, Feeding 3.2V from an external supply  into the VCC end of NT2, and 1.6V from an AA battery into the +1V8 end of NT1 
with the board traces still cut, and lo and behold, the HackRF actually was brought back to life. 

Nothing is now getting hot and I can listen to my local FM stations using CubicSDR. I found this simply amazing!
The received felt a bit deaf though and turning on the preamp just increase the nouse level, but that was with the RF screen
unmounted. Putting it back returned the receiver back to normal. I also tested transmitting a CW signal which also worked.

So it seems like the only thing that broke was the regulator alternatively the capacitors C126 and/or C127 are now more or less shorted.

Further testing revealed that the 1.8V part of the regulator is still working. Instead of feeding 1.6V from a battery and turning off Q5,
disabling the 3.2V part of the regulator by shorting pin 2 of R52 to ground made it possible to get 1.8V from the regulator whenever 3.2V
is fed into the VCC. This way the HackRF still works and nothing is getting hot.

The plan is now to put it back into operation by feeding 3.3V from the nearby RaspBerry PI 4 which also connects to the USB. 

:-)

It would be interesting to know if there is any testing program available, e.g. using hackrf_debug.

-- Christer