greatscottgadgets/hackrf #669: amplifier question

luoGanda, 2019-11-28:
This is not an issue, more like a question or tehnical support.

It seems that amplifier on my hrf doesn't work any more.
When i turn it on - for example in any sdr software, signal is much worse than without it.
Does this mean that amp has gone to heaven?
Is there some trick to check/correct this or something?
Could it be because of mistakenly using a bias voltage without using an external amp?

miek, 2019-11-28:
Yeah, if the signal level reduces when the amplifier is turned on then that is a sign that the amplifier has failed. It won't have been caused by enabling the bias voltage, it's usually caused by a strong signal nearby.
If you got your HackRF from an official reseller, you can get in touch with them to get a replacement unit.

febs, 2019-11-28:
This failure of the amp is reported quite often and I am quite afraid of it happening, then I ask:
_how_strong_ a signal must be to screw the amp?

What could be a practical example of that? Like, a 5 watt UHF handeld radio trasmitting in the same room could blow it?

To have an idea.

Thank you,

luoGanda, 2019-11-28:
aha, i see..
Reseller thing is out of the question, because warranty is over, but i might fix it somehow, altough i am not much into electronics.

Hmm, then technically - if someone want's to screew the amp anywhere(even in audio amps) - just overdriving it with strong enough signal is enough?

miek, 2019-12-02:
> This failure of the amp is reported quite often and I am quite afraid of it happening, then I ask:
> _how_strong_ a signal must be to screw the amp?

The safe input power for all configurations is -5dBm (from https://github.com/mossmann/hackrf/wiki/HackRF-One).
> 
> What could be a practical example of that? Like, a 5 watt UHF handeld radio trasmitting in the same room could blow it?

To get an idea, you can calculate the free space path loss to see how far away the transmitter should be. There are online calculators such as https://www.everythingrf.com/rf-calculators/free-space-path-loss-calculator. 5W is about 37dBm, so we need 42dB of path loss to get to that safe -5dBm value. Assuming 433MHz & antennas with no gain, the calculator above says that the 5W transmitter should be at least 7m away.

febs, 2019-12-02:
Thank you, it is very clear.
I probably have screwed my unit then.

miek, 2019-12-02:
> Reseller thing is out of the question, because warranty is over, but i might fix it somehow, altough i am not much into electronics. Anyone knows what component exactly to replace for this?

@luoGanda We do our best to fix/replace faulty or failed units and there's no formal expiry on that. We're can definitely replace your unit - if you're having any issues organising that with a reseller, please send any details to info@greatscottgadgets.com

miek, 2019-12-02:
> Thank you, it is very clear.
> I probably have screwed my unit then.

@febs The numbers above have some safety margin (the safe level is higher if the RF amp is disabled) & are theoretical - your HackRF may still be fine even if you have transmitted closer. It's simple to check: if the received signal level drops when you enable the RF amplifier, then it has probably failed. If you think you do have that issue, please open up a new issue with the details and we can get it sorted out.

luoGanda, 2019-12-07:
@miek: replacement is probably out of the question, because i have a clone of hrf, not the original one.

But, if replacement for this is out of the question,
if anyone knows the exact component to replace inside the hrf(which has amp in), then this info is welcome.

miek, 2019-12-07:
Ah OK. The RX amplifier is labelled U13 in the KiCad files, and the TX amp is U25. The part number is MGA-81563.

luoGanda, 2019-12-07:
aha - i forgot that there are those files in source code..

closed 2019-12-08 by miek