greatscottgadgets/hackrf #670: GRC + HackRF One: Noise in a simple FM signal while transmitting

m-anish, 2019-12-05:
### Steps to reproduce
1. Please see the [thread on qrz here](https://forums.qrz.com/index.php?threads/hackrf-test-broadcasting-fm-silence.683253/).

### Expected behaviour
I expected a low noise floor on the produced FM signal, but that wasn't the case.

### Actual behaviour
I would expect a low noise floor on a hackRF when producing FM signals (certainly lower than a cheap chinese FM transmitter), but the noise levels are uncomfortably appreciable. 

### Version information
Ubuntu 18.04 running GRC on a thinkpad T440p with a dual core i5 processor.

**hackrf_info output:**
```
hackrf_info version: unknown
libhackrf version: unknown (0.5)
Found HackRF
Index: 0
Serial number: 0000000000000000674068dc38a1881f
Board ID Number: 2 (HackRF One)
Firmware Version: 2018.01.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x00684353
```

GRC version 3.7.11

I would like advice on how to investigate this further. I have a 100MHz oscilloscope handy if that helps.

straithe, 2021-02-28:
I'm going to close this as there hasn't been a response in a while, but please re-open this issue or open a new one if you still need assistance.

closed 2021-02-28 by straithe

m-anish, 2021-03-04:
Sure. Last I checked with recent software versions, it seemed okay to me. I'll comment again if I investigate this more deeply and face the issue.