greatscottgadgets/hackrf #673: Captures taken with timestamps reveal gaps... Host/USB/HackRF problem?

mgesteiro, 2019-12-11:
### Steps to reproduce
1. Patch hackrf-transfer.c to include lightweight timestamps
2. Capture at 20Ms some signal at 1Ghz
3. Extract time-stamps and analyze their behaviour

### Expected behaviour
All time-stamps should be constant (at least in average)

### Actual behaviour
Time stamps are not constant and big gaps appear scattered (some as big as 240 milllisecs).

### Version information
**Ubuntu 18.04 - Intel i7-16GiB / Raspian Buster - Raspberry Pi 4-4GiB**:

**hackrf_info version: unknown
libhackrf version: unknown (0.5)
Found HackRF
Index: 0
Serial number: 000000000000000087c867dc2940bb5f
Board ID Number: 2 (HackRF One)
Firmware Version: 2018.01.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x0069474e**

### Output
```
./hackrf_transfer -S 20971520 -r - -f 1030000000 -s 20000000 -a 0 -l 32 -g 6 > IQ.raw 2> IQ.log
```

### Additional information
In the begining we where capturing to an **USB-3 ME hard disk**, but at the end we were using **RAMdisks to discard an storage issue**. We tried also buffering (-S). We got some improvements but still see notable variations in the timestamps.

We are trying to understand this behaviour, if samples are lost, if the USB bus is the problem, if the storage...

**Any advice?**

miek, 2019-12-11:
Could you push up your changes for adding timestamps (or post the patches)? Then I can try and reproduce the problem.

mgesteiro, 2019-12-11:
Sure. Here they are:
https://github.com/mgesteiro/hackrf/commit/247fc85598d64b79714ba6529e62956d7244a82f

miek, 2019-12-16:
Aside from checking the averages over time, adding timestamps on the host isn't going to give a reliable result. There's no guarantee that each USB transfer will be handled immediately, and there are large buffers available so you may see an occasional long delay in normal operation.

Ideally it would be tested with timestamps added in the HackRF firmware, but that gets a bit more complicated. You could try editing the firmware to replace the first sample in a transfer with an incrementing ID, and see if there are any gaps.

mgesteiro, 2019-12-17:
> Aside from checking the averages over time, adding timestamps on the host isn't going to give a reliable result. There's no guarantee that each USB transfer will be handled immediately, and there are large buffers available so you may see an occasional long delay in normal operation.

We thought about that too, and we looked specifically for those delays: at lower sample rates, the buffers almost compensate the time drift, but the higher the sample rate, the bigger the drift in long captures (~10 min.). 
[image]

> 
> Ideally it would be tested with timestamps added in the HackRF firmware, but that gets a bit more complicated. You could try editing the firmware to replace the first sample in a transfer with an incrementing ID, and see if there are any gaps.

It's a clever idea, but I'm not sure if it will help: if there are gaps in the samples, it will be clear, but if the gaps are in time (drifting, as I mentioned above), we couldn't notice.

We are preparing a "black box" experiment, with a controlled source signal to see what we get at the output: if anything useful comes out we'll tell.

In the meanwhile, any other ideas/suggestions?

Thanks and best regards

miek, 2021-04-01:
Closing this as we haven't been able to reproduce this & there's been no update in a while. Feel free to reopen if some new info comes up though!

closed 2021-04-01 by miek