greatscottgadgets/hackrf #674: RX question:The envelope of the received signal is sinusoidal.It's caused by no Frequency offset cancellation?

FamerWaitingforRain, 2019-12-16:
### Steps to reproduce
1.hackrf one transmit the continuous square wave signal in 433MHz

2.other hackrf one receive the above signal through antenna

3.Theoretically，rx will get the continuous square wave,but the envelope of the received signal is sin.

### Expected behaviour
Theoretically，rx will get the continuous square wave,the amplitude of signal is +-amp
[image]

### Actual behaviour
I get the square wave ,but the amplitude envelope of the received signal is sin.
[image]
[image]

### Version information
**Operating system**:
ubuntu 16.04
**hackrf_info output:**
Found HackRF
Index: 0
Serial number: 0000000000000000901868dc2429b48b
Board ID Number: 2 (HackRF One)
Firmware Version: 2018.01.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x00594358

Found HackRF
Index: 0
Serial number: 0000000000000000901868dc241fae8b
Board ID Number: 2 (HackRF One)
Firmware Version: 2018.01.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x0062475e

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.
gnuradio 
### Output
```

```

FamerWaitingforRain, 2019-12-16:
Expected:
[image]
Actual:
[image]
[image]

kpreid, 2019-12-16:
HackRF (and other SDRs) deal in complex-valued (also known as quadrature or IQ) signals. Any frequency offset (and there will always be a frequency offset in non-ideal physical equipment) causes a continuous rotation of the phase of the received signal. If your transmitted signal has two symbols (levels/values) and you are looking at only one component (I or Q) of the received signal, then a sinusoidal amplitude variation (and inversion) is to be expected.

In order to receive such a signal in a form similar to the original, you need to modify your receiver to detect and synchronize with the phase of the transmitted signal. GNU Radio has blocks that can do this, but I do not have expertise to recommend one to you for this particular application.

LiuYu0809, 2019-12-17:
> HackRF (and other SDRs) deal in complex-valued (also known as quadrature or IQ) signals. Any frequency offset (and there will always be a frequency offset in non-ideal physical equipment) causes a continuous rotation of the phase of the received signal. If your transmitted signal has two symbols (levels/values) and you are looking at only one component (I or Q) of the received signal, then a sinusoidal amplitude variation (and inversion) is to be expected, because when the phase has shifted by 90° the signal content is now mostly in the other component — the cosine instead of the sine, essentially.
> 
> In order to receive such a signal in a form similar to the original, you need to modify your receiver to detect and synchronize with the phase of the transmitted signal. GNU Radio has blocks that can do this, but I do not have expertise to recommend one to you for this particular application.

Thanks for your reply sincerely.
However, I still have some questions about this. The frequency offset exist in the received signal inevitably,I think the reason is the difference between TX's CLK and RX's CLK.Therefore when the received signal is multiplied by the RX Local oscillator frequency to downconvert，the received signal will downconvert to a frequency difference(flo-fc) instead of baseband.
What confused me is the relationship between received signal's sinusoidal amplitude variation and I&Q signal. 
My transmit signal's I&Q baseband is the same both in amplitude and phase.And the difference between I RF signal and Q RF signal is the phase.why only look at the I or Q of received signal,will the amplitude change?

kpreid, 2019-12-17:
> My transmit signal's I&Q baseband is the same both in amplitude and phase.

If a signal's I component is the same as its Q component, then on a constellation plot (I vs. Q graph) it can be visualized as a diagonal line — phase of 45°, because it's just like plotting the elementary function "y = x". Rotate (phase shift) this by +45° and it's a vertical line, only in the Q component. Rotate this by -45° and it's a horizontal line, only in the I component.

This is a general fact about objects in coordinate systems: if you rotate a flat object, from some angles it disappears. That's what you're seeing here. In order to always be able to see it, you must look at it from more directions, or from the correct (phase matched) direction.

Try using the constellation plot available in GNU Radio. If you attach it to the transmitter's input signal you will see the 45° line segment, and if you attach it to the receiver you will see a rotating line segment of constant amplitude.

---

Note also that if you were to change the transmitted signal so that I and Q were not identical (making it no longer "flat"), you would no longer see the amplitude dropping to zero. But it is still the case that by looking only at received I or only at received Q, you are not seeing all the information of the signal — you're seeing some linear combination of the transmitted I and the transmitted Q.

closed 2020-01-06 by miek

varuliant, 2020-12-20:
Hi, I know this discussion is closed. But, I wonder, what is the solution? 
I am a newbie in GNU Radio and have similar problem
Thanks in advance

elmaghba, 2023-01-13:
@kpreid  Thank you for the great insight! Could you please refer to some resources that provide the math behind this behavior (freq offset)? 

Really appreciated!