greatscottgadgets/hackrf #676: Multiple hackr devices receive signals at the same time.

xtu20, 2019-12-17:
Can one host run two hackrf devices at the same time?I plan to use two hackrf devices to scan different frequency points at the same time, but I can't find the method at present.

miek, 2019-12-17:
Yep, one host can run multiple HackRFs. To select the device, both `hackrf_transfer` and `hackrf_sweep` take the `-d` argument to specify the serial number of the desired HackRF.

xtu20, 2019-12-18:
> 是的，一台主机可以运行多个HackRF。要选择设备，请选择两者`hackrf_transfer`并`hackrf_sweep`接受`-d`参数以指定所需的HackRF的序列号。

Thanks for your advice. I have tried successfully, but I want to know how many hackrf can run at most at the same time. I can test 3 at present, thanks again.

miek, 2019-12-18:
There's no limit in software, but it's hard to say how many will actually work without trying it; eventually it would be limited by the host computer's ability to handle that much data coming in.

closed 2019-12-23 by miek