greatscottgadgets/hackrf #677: Can it transmit?

lwdallas, 2019-12-17:
Great Scott Gadgets asked me to post this here:

Before I go in blind, my use case is that I use Mac and Ubuntu(elementaryOS). I would like to receive and transmit ham bands including HF when I travel. 

What should be one my wish list?

miek, 2019-12-17:
The HackRF One does transmit but it may not be suitable for use as a ham radio transceiver out of the box (depending on your needs): the transmit power is 5dBm to 15dBm (3.2mW to 32mW) in those bands. If you wanted to transmit with more power then it would need external filtering and amplification, so it becomes a bit more of a project.

lwdallas, 2019-12-17:
I'm game if you can help me build a wish list!

miek, 2019-12-23:
Roughly: you'd need a bank of band-pass filters on the HackRF output, then an amplifier, then another bank of (probably low-pass) filters after the amp.

I don't know of anyone who has put together a setup like this, so I can't give any more detail than that - you'd have to do some more research and testing to come up with a good setup.

lwdallas, 2019-12-25:
Agreed. Thanks for your help!

Sincerely,
Lonnie Webb

closed 2019-12-25 by miek