greatscottgadgets/hackrf #680: Not receiving MacBook Pro

mattmac553, 2019-12-22:
### Steps to reproduce
1. Plug HackRF into usb a to C
2. Open gqrx 

[image]

### Expected behavior
Show normal somewhat flat line

### Actual behaviour
Get a weird hump in the display

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
```
Insert any commandline or build output here
```
Only happens on MacBook Pro not my older windows computer

mattmac553, 2019-12-22:
There should be pleanty Of p25 control channels and voice channels being displayed

miek, 2019-12-23:
The "weird hump" that you see is probably just from the anti-aliasing filter - it's usually set to a value just under the sample rate but can be made narrower for better attenuation of out-of-band signals. You can adjust the width of it with the "bandwidth" setting in GQRX.

closed 2019-12-24 by mattmac553