greatscottgadgets/hackrf #681: broken USB?

JonaldJohnston, 2019-12-23:
Hello!
just yesterday, i was going to try to flash new PortaPackfirmware onto my HackRF has recently decided to not talk with my laptop or PC anymore, in normal and DFU mode as well. 
here's what I've found:

The hackrf still powers up fine and Portapack functions as normal
dfu-util does not flash or list any devices when the hackrf is in dfu
lsusb does not show anything related to the hackrf in normal and dfu mode
dmesg does not change when i plug or unplug the hackrf in normal and dfu mode

I have found the above results with 6(!) separate micro USB cables, including some that i know work with the hackrf, and on two separate computers. This leads me to believe its an issue with the micro usb connector, anything i can do to fix it?
Thanks!

JonaldJohnston, 2020-01-01:
bumping this, but i also want to know if there is a way i can program the hackrf without the usb? i don't really need to transfer data over the usb if i'm just using it for portapack, and it powers through the port just fine.

eeprommemory, 2020-01-05:
same problem portapack firmware has a bug kills usb

JonaldJohnston, 2020-01-08:
so this is not a hardware issue?
how do i reflash firmware without a usb connection though?

eeprommemory, 2020-01-08:
at first look i thought it was a software issue but my hardware is damaged.
check #684 on how to test your usb port.
i used mine as a hackrf for years no problem plug it in to a battery bank once and end of fun.

JonaldJohnston, 2020-01-09:
fun thing is that my usb connection seems to be fine, no short to ground

eeprommemory, 2020-01-09:
https://www.nxp.com/docs/en/data-sheet/LPC4350_30_20_10.pdf
page 139 has an interesting note on it.
***************
Remark: Applying 5 V to the USBn_VBUS pins for a short time while the regulator ramps up might compromise the long-term reliability of the part but does not affect its function.
***************
so what happens if the regulator take a real long time to ramp up.
all my problems started after adding the portapack.
i got 5 years out of it then blow it in the first week of having the portapack.

vasja2000, 2020-01-09:
Такая же проблема, только у меня portapack h2 (

vasja2000, 2020-01-09:
same problem, only I have portapack h2(((

vasja2000, 2020-01-11:
I think this was due to the fact that the firmware was not installed correctly. Stitched on this manual, https://www.youtube.com/watch?v=5G5blPPIIBo. at the moment of check connected hackrf+portapack to power bank. On portapack h2 + hackrf the white screen lit up, the firmware did not load, further pressed reset. After this procedure, kali does not see hackrf!(((Whatever I do. Can you give instructions on how to try resuscitating the usb port on hacrf?

vasja2000, 2020-01-11:
the dmesg-w ### command gives the answer:
[  407.058438] usb 1-9: new full-speed USB device number 4 using xhci_hcd
Provided: 
1. connect to usb 2.0
2. the dfu button is pressed, I connect the cable, the DFU button is pressed
3. Hackrf enters DFU mode ( only 3v3 is lit).
the hackrf_info ### command gives the answer:
hackrf_info version: git-e9c6c2d
libhackrf version: git-e9c6c2d (0.5)
No HackRF boards found.

Is it possible to physically open the usb port on the Board itself (jump the legs of the chip) so that data flows?

jboone, 2020-01-11:
If you connect the USB cable while the DFU button is pressed, the HackRF will be in DFU (device firmware upgrade) mode. It will not act like a HackRF, and will not respond to hackrf_* commands. You can only perform a DFU load of the firmware, which is done with dfu-util. Please see the Wiki for more information: https://github.com/mossmann/hackrf/wiki/Updating-Firmware

JonaldJohnston, 2020-01-14:
i am using dfu-util and it does not show anything, dmesg doesn't change either

miek, 2020-03-24:
Moved to #715

closed 2020-03-24 by miek

vasja2000, 2020-04-03:
hi! Colleagues, help me find the scheme exactly Portapack_h2. We need it.

rob-ltx, 2020-10-10:
I too now have a HackRF One that has a USB problem.  I have tried three desktop PCs, several cables but normally use 
the one supplied with it.  Switching to DFU mode  shows nothing with lsusb or dmesg either.
Physical examination of U15 and components in that area under a microscope shows nothing unusual so it appears to have died for no reason.  Nothing associated with its use as a receiver monitoring the ISS has changed but on switching on the system last Sunday there were no signals and on investigation no USB connection.
Very disappointing for a device costing over £200 when a Funcube Dongle Pro+, a LimeSDR Mini and a RTL-SDR Dongle all work quite happily on the same PC. :-(

josmargoncal, 2021-02-14:
I have the same problem, I try all and nothing

stutteringp0et, 2021-09-15:
Same problem - inspected the diode specified in issue 715 and it looks fine.  No DFU mode either.

straithe, 2021-09-15:
Can you please open new GitHub issues for the errors you are encountering?

Commander-Crash, 2021-09-18:
same issue. Wish there was a way to flash it.

goa, 2022-02-01:
Same issue here.

straithe, 2022-02-01:
Can each of you please let me know:
* What portapack version you have
* What HackRF version you have
* If you have tried DFU mode without the portapack on the HackRF? 

We at GSG are doing a deeper dive into this problem and would appreciate the information!

goa, 2022-02-01:
Thank you for your help @straithe.

>     * What portapack version you have
I have a portapack H1

>     * What HackRF version you have
Not really sure how I can determine my HackRF version without having a USB connection to the computer, but I have `Mayhem v1.4.3` so I suppose the HackRF version must be `2021.03.1`.

>     * If you have tried DFU mode without the portapack on the HackRF?
I have, but it is still not recoverable as any type of USB device by MacOS or Windows 10 (two different Windows 10 computers).

kreativdax, 2022-07-20:
I have the same problem is there any update to this problem????

kreativdax, 2022-07-23:
Any update on this I have same problem still too

straithe, 2022-07-23:
I suggest opening up a new issue.

AndyH76, 2024-12-11:
Has anyone solved the USB problem? The same problem on my HackRF+Portapack H1

goa, 2024-12-11:
Still nothing on my side @AndyH76 ....

AndyH76, 2024-12-11:
> Can each of you please let me know:
> 
> * What portapack version you have
> * What HackRF version you have
> * If you have tried DFU mode without the portapack on the HackRF?
> 
> We at GSG are doing a deeper dive into this problem and would appreciate the information!

Did GSG figure out what the problem is?

goa, 2024-12-11:
> > Can each of you please let me know:
> > 
> > * What portapack version you have
> > * What HackRF version you have
> > * If you have tried DFU mode without the portapack on the HackRF?
> > 
> > We at GSG are doing a deeper dive into this problem and would appreciate the information!
> 
> Did GSG figure out what the problem is?

nope

eeprommemory, 2024-12-12:
replace CPU U23 change R62 to 100 ohm R58 to 27 ohm R57 to 27 ohm.
mine still works after mods so all seems good.

eeprommemory, 2024-12-12:
multi meter in diode test mode and test to see if R57 or R58 is short to ground and post the results.

AndyH76, 2024-12-12:
> multi meter in diode test mode and test to see if R57 or R58 is short to ground and post the results.

My HackRF+Portapack H1 works completely. The problem is only with the USB port. First of all, I checked the resistance on the USB connector contacts. There is no short circuit. Between ground and D+ D- 1.7 MOhm. When connecting the power, the current in idle mode is ~180 mA. There are no visually faulty elements. I will try to remove U15. I do not think that the problem can be in U23.

straithe, 2024-12-13:
> Has anyone solved the USB problem? The same problem on my HackRF+Portapack H1 

For issues with HackRF+Portapack combos, please contact the company you purchased from. It is likely the HackRF in the combo is a clone and don't offer support for external hardware. If you did order a HackRF One from a GSG authorized reseller, please let us know.

straithe, 2024-12-13:
> > Can each of you please let me know:
> > 
> > * What portapack version you have
> > * What HackRF version you have
> > * If you have tried DFU mode without the portapack on the HackRF?
> > 
> > We at GSG are doing a deeper dive into this problem and would appreciate the information!
> 
> Did GSG figure out what the problem is?

Each issue can vary greatly depending on operating system, virtual machine use, cable used, etc. If you would like support please open your own issue and we will start the technical support process.