greatscottgadgets/hackrf #682: GNURadio 3.8 Companion use with HackRF One Device other than osmocom?

Cingulara, 2019-12-24:
### Steps to reproduce
1. setup and run GNURadio 3.8
2. Osmocom block not available
3. https://github.com/gnuradio/gnuradio/issues/2838 states Osmocom won't work with 3.8

### Expected behaviour
I see from my own use and https://github.com/gnuradio/gnuradio/issues/2838 that the osmocom that I use in GNURadio Companion 3.7 does not work right with 3.8. Is there another block I can use with the HackRF One in 3.8? 

### Actual behaviour
You get a Missiong Block opening 3.8 with an older Osmosdr_sink module setup. 3.7 works fine. 

### Version information
OS is the latest Ubuntu Desktop

**hackrf_info output:**
```
$ hackrf_info
hackrf_info version: unknown
libhackrf version: unknown (0.5)
Found HackRF
Index: 0
Serial number: 000000000000000087c867dc2b265c5f
Board ID Number: 2 (HackRF One)
Firmware Version: 2018.01.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x00604756
```

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.
GNURadio Companion 3.8

### Output
[image]

miek, 2019-12-24:
You can use the SoapySDR interface instead.

Cingulara, 2019-12-24:
That works thank you! I set that up for GNURadio. I also saw the Pothos flow editor as well.

closed 2019-12-24 by Cingulara

chrislafave, 2020-03-12:
I love you, miek. Seven words can make a huge difference. I should have looked sooner for an alternative to Osmosdr "blocks" for using HackRF in GnuRadio-Companion. I would have saved myself a _lot_ of hair-pulling grief (trying to downgrade GRC to 3.7 and still get everything necessary to compile, etc). Also, I didn't know that there were actual alternatives to GRC (I'm looking right now at the Pothos project's dataflow framework and it's beautiful)

PureRockets, 2021-06-17:
Can someone provide specific directions to install the SoapySDR interface on Ubuntu 18.04 with a HackRF? It seems it too has a dependency on osmocom? https://github.com/gnuradio/gnuradio/issues/2838